jandelgado/jled #5: 10-bit doesn't appear to work on UserFuncs

JayElDubya, 2018-09-17:
I scratched my head for awhile until I figured it out but it appears the max brightness the UserFunc can support is 255. I'm using a Wemos D1 mini pro (pretty common esp8266 board) and the pwm brightness level is 10bit. I was trying to write a custom UserFunc that would blink the onboard LED between off and a lower maximum brightness. However, it's an active-low LED so I need to set it to 1023 for it to be off which I cannot do with the library expecting uint8_t. 

I read that you added 10-bit support but I'm not sure how to activate it in Platform.io. My platform.ini file for my project contains:

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino

Has the UserFunc code also been modified to support 10-bit? If so, can you point me in the direction to get going? If i create a function and just have it return 1023, the LED lights up the same as if I did a return 255 (because it's a BYTE and hence clips it to 255). 

Thanks for the great library!

-Josh

jandelgado, 2018-09-22:
Hi Josh,

internally the library only uses 8 bits and assumes 0==off and 255=full brightness/on. 

When compiled for ESP8266 (ESP8266 should be #defined) the value is scaled to 10 bits before the the actual analogWrite(). So it is transparent for the application.

For an low-active LED just use LowActive() during construction: 

JLed led = JLed(9).Breathe(1500).LowActive();

This should do the trick.

To sum up: Return 0 from your UserFunc to turn the LED off and 255 for full brightness. If the LED is low-active, construct it with LowActive()

greetings Jan

jandelgado, 2018-10-03:
Closed as not being a bug.

closed 2018-10-03 by jandelgado