jandelgado/jled #6: Feature request:  Return status from Update()

sslupsky, 2018-09-23:
I have a use case where it would be helpful to know if an LED sequence is queued so that I can stay awake until it is complete.  Looking at your code is seems to me that if you returned a boolean status from the Update() method, that would likely do the trick.

I made some changes and it appears to work but not sure if I have covered all the corner cases with my testing.  Please let me know if you would like a PR.

jandelgado, 2018-09-23:
Sounds like a reasonable enhancement and should be easy to do. Did you also add some Unit-Tests? Nevertheless, make a PR, I'll check it. Thanks Jan

jandelgado, 2018-09-28:
I added the feature in the 2.4.0 release

closed 2018-09-28 by jandelgado