jandelgado/jled #7: [Question]: Looking for a function like myLed.isOn(), myLed.isOff()

OllisGit, 2018-10-06:
First: I like your lib very much! I appreciate your effort.

What is the best way to check if a led is currently on/off or (optional) get the brigthness value?

Thx
Olli

jandelgado, 2018-10-06:
Currently there is no direct way to check the status of the LED. What is your concrete use-case for that feature?
regards
Jan

OllisGit, 2018-10-07:
I am using different LEDs for indicating several events. 
The LEDController function is running on a seperate Thread/TaskHandle.
Each time a event occours in the main-thread a differnt LED is switched on (with/without animation). via LEDController.
At the moment I am "storing" the  event-information in a lot of fields in led controller. And these fields "pollute" my LEDController-Code ;-)

It would be great if i have the opportunity to ask leds which state they have. 
E.g. "emergency-event: if orange led is on, turn off, turn on red led"

BR
Olli

jandelgado, 2018-10-11:
Currently you can check if an effect is running by looking at the return code of the Update() method. In the next release I'll also add a method to check if the effect is done or not (e.g. led.IsRunning())
Would that solve your problem?

OllisGit, 2018-10-12:
Sounds good!
I think I can store the returned Update-Flag from each LED and provide my own status-methode (as a workaround, waiting for next release ;-).

Thx, in advance
Olli

closed 2018-10-19 by jandelgado

jandelgado, 2019-01-31:
The 4.0.0 release of JLed (see version in master branch) now has a IsRunning method:

bool JLed::IsRunning() const

indicating if the effect is still running.