jandelgado/jled #11: [Question] Best way to change "animation"

OllisGit, 2018-10-15:
Example: 
I have only one LED. If the programm runs "smoothly" the led has the animation "Breathe()".
But if an error occours (wifi-lost) then the led should switch to fast blinking.

At the moment my implementation looks like this. Everytime I need to switch the animation, I create a new JLED-Instance. It works, but maybe there is a more "elegant" solution. Maybe related to Issue #10 

```
JLed leds[] = {
		JLed(BUILTIN_LED).Breathe(2000).Forever()
};

loop(){
       leds[0].Update();
}

errorOccoured(){
     leds[0] = JLed(BUILTIN_LED).Blink(200, 200).Forever();
}

```

jacfontaine, 2018-10-15:
Hello,

There is no need to redefine the object each time.
Just call the variable with the new parameters.

`ex: leds[0].Blink(50, 1800).DelayBefore(450).Forever();`

Regards,

Jacques

OllisGit, 2018-10-17:
Thx, again for your fast answer.
I changed my code and it works flawlessly.

closed 2018-10-17 by OllisGit