jandelgado/jled #12: How do I set up conditionals?

fjnorb, 2018-11-04:
I am trying to use your library for the first time for non-blocking LED effects in a Teensy 3.5 sketch. Your examples work fine, however they don't demonstrate how to blink an LED in response to a conditional, such as a button press.

In this case I would like to blink an LED fast 20 times when a button is pressed.

I have tried numerous things, and I think that this should work but it doesn't:

```
#include <Bounce.h>
#include <jled.h>

const int ledBell1 = 2;
Bounce button0 = Bounce(23, 5); // 5 ms debounce time
JLed jledBell1 = JLed(ledBell1);

void setup() {
}

void loop() {
  button0.update();
  jledBell1.Update();

  if (button0.risingEdge()) {
    JLed jledBell1 = JLed(2).Blink(100,100).Repeat(20);
  }
}
```

I have also tried adding a second jledBell1.Update(); after the re-declaration n the if block, and it still doesn't work. Occasionally the LED will come on solid. Serial.print'ing the return value from Update() returns either 0 or 1 depending on if the LED is on or off, but the LED never blinks. Can you help me understand why this doesn't work? I can't imagine this isn't supported in the library as it would make it fairly useless. Thank you.

jandelgado, 2018-11-04:
Hi, 
you are defining a new `JLed` object, which exists just in the scope of the if-block, it hides the previous definition. Just change your code, so that the `jledBell1` objects gets updated on the keypress:

```
...
if (button0.risingEdge()) {
    jledBell1 = JLed(2).Blink(100,100).Repeat(20);
  }
...
```

greets
Jan

fjnorb, 2018-11-05:
It's so obvious now that you show it. Thank you!

closed 2018-11-11 by jandelgado