jandelgado/jled #16: ESP8266: Watchdog Timer Reset when using v4.0.0-rc0 release

marrold, 2019-02-16:
Hi, I'm trying to use a basic example on a NodeMCU ESP8266 board, but I am getting Watchdog Timer Resets.

Error:

```
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Code:

```
#include <Arduino.h>

#include <jled.h>

JLed led = JLed(LED_BUILTIN).On().DelayBefore(2000);

void setup() {
}

void loop() {
  led.Update();
}
```

Am I doing something wrong?

Thanks!

jandelgado, 2019-02-16:
I can reproduce the problem and am working on a fix - thanks for the report.

jandelgado, 2019-02-17:
I located the problem in the newly introduced placement new, which needs to use an aligned buffer on the ESP8266. I'll provide a fix tomorrow.

closed 2019-02-17 by jandelgado

marrold, 2019-02-18:
This is working perfectly, thanks

PR #17 (2019-02-17): ESP8266 needs DWORD aligned byte buffer for placement new.
fixes #16