jandelgado/jled #19: Question: Changing sequence

marrold, 2019-02-18:
I suspect that I am showing my lack of c++ knowledge here so apologies in advance. 

How can I update my sequence to a new sequence whilst my program is running?

```
JLed standard[] = {
    JLed(5).On().DelayAfter(2000),
    JLed(4).On().DelayAfter(1000),
    JLed(5).Off(),
    JLed(4).Off(), 
    JLed(14).On().DelayAfter(2000),
    JLed(14).Off(),
    JLed(4).On().DelayAfter(1000),
    JLed(4).Off(),    
};

JLed pelican[] = {
    JLed(14).Blink(500, 500).Repeat(3),
    JLed(5).Blink(500, 500).Repeat(3),
};

JLedSequence sequence(JLedSequence::eMode::SEQUENCE, standard);
```

Thanks

jandelgado, 2019-02-18:
Currently this is not supported, but you could create two different  `JLedSequence` objects, e.g. `sequence1` and `sequence2` and mutually call `Update()` on just one of these.

marrold, 2019-02-18:
I figured this would be the work around - Thanks!

closed 2019-02-18 by marrold