jandelgado/jled #20: Not working with Sandeep's Lora Library

heinduplessis, 2019-04-22:
This library seems to be incompatible with Sandeep's Lora library - https://github.com/sandeepmistry/arduino-LoRa

As soon as I use the Led functions I no longer receive message on Lora. Might be an interrupt thing?

jandelgado, 2019-04-23:
JLed does not use interrupts, it relies on an `Update()` method being called from the main event `loop()` function. Besides that, only `pinMode()` and `analogWrite()` are used. No hidden magic.

Is it possible that you are using GPIO pins with JLed that are already in use by LoRa hardware? Try changing the pins the LEDs are connected to.
  
In general this defect report is too generic as to be accepted. Please provide further details like hardware used, test setup, example code etc. so there is a chance of getting the observed behaviour reproduced.

heinduplessis, 2019-04-24:
Thanks, I'll refine my tests and see if I can distil the essence of the problem! I'll re-open if needed.

closed 2019-04-24 by heinduplessis