jandelgado/jled #21: version is incorrect in library.properties

sslupsky, 2019-05-07:
This line is not compliant with the spec for library.properties version property:

https://github.com/jandelgado/jled/blob/6643a5f4122dbf3a284664564a98faecc0d85c65/library.properties#L2

It is throwing an "Invalid Version" error at compile time.

I think you need to remove the "v" at the beginning.

jandelgado, 2019-05-07:
Strange - that seemed to be a new behaviour, but your are right, thats what the [spec ](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) states:

> version - version of the library. Version should be semver compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid

Always had the "v" in front of the version before. I'll soon change it and release a new version.

jandelgado, 2019-05-07:
I just released 4.1.1 does it work now?

sslupsky, 2019-05-07:
That fixed it!  Thank you!

sslupsky, 2019-05-07:
I think it might be an issue introduced in one of the recent IDE updates?  Not sure.  I found this reference:

https://forum.arduino.cc/index.php?topic=596113.0

Which referred to the spec:

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Which refers to "semver".

jandelgado, 2019-05-08:
Excellent - thanks for pointing out the problem

closed 2019-05-08 by jandelgado