jandelgado/jled #22: sequence.ino example doesn't work on ESP32

iZhangHui, 2019-05-08:
It seems that just the last sequence effect work.

jandelgado, 2019-05-08:
Hi, could you please describe a little bit more detailed the expected and observed behaviour?

The last "effect" in the [sequence](https://github.com/jandelgado/jled/blob/master/examples/sequence/sequence.ino) example is to turn the LED off (i.e. `Off()`). 

https://github.com/jandelgado/jled/blob/8701e212660f497e6b0cb21a8fb0a0f8c85e8e07/examples/sequence/sequence.ino#L10

Regarding your report, that would mean that the the whole demo would not show anything, correct?

iZhangHui, 2019-05-08:
Yes, your describe is right, the whole demo would not show anything.
When I comment the codes of Line 11, 12, 13, the demo will fadeoff 3 times, and so forth.

jandelgado, 2019-05-08:
Ok, will check this in a few days. Thanks for the report.

jandelgado, 2019-05-08:
Good news: I've spotted the problem in the ESP32 HAL, in the pin-channel assignment. Will fix it coming weekend.

jandelgado, 2019-05-11:
@iZhangHui release 4.1.2 should fix the problem - could you please retest?

iZhangHui, 2019-05-11:
I will retest and give you feedback.

iZhangHui, 2019-05-12:
Verified, It works.  :）

closed 2019-05-12 by iZhangHui

jandelgado, 2019-05-12:
Excellent - thanks.

PR #23 (2019-05-11): Issue22 fix esp32 channel assignment
fixes #22 
tested successfully with heltec wifi lora esp32
commit b6f2cd86 (2019-05-11): fix esp32 channel assignment (#22)

* fixes #22 (esp32 led pin-to-chan assignment)
* led pin configurable in sequence example