jandelgado/jled #25: [Question] Static on and off

iZhangHui, 2019-06-18:
Hi @jandelgado , please help me to check the code below.  Why  led.Off().Update() won't turn off the LED? Thank you.

`#include <jled.h>

JLed led(LED_BUILTIN);

void setup() {
  led.On().Update();
  delay(2000);
  led.Off().Update();
  delay(2000);
}

void loop() {
  led.Update();
}`

jandelgado, 2019-06-19:
Hi, what Board are you using? I remember that for example I had a NodeMCU (ESP8266) board which had the builtin LED connected to VCC so that it was low active. So in your example you perhaps just see `On()` and `Off()` reversed. If this is the case, try to construct your led object like `auto led = Jled(LED_BUILTIN).LowActive();` which will invert the output.

br
jan

jandelgado, 2019-06-20:
I checked your example and had the same problem ... In addition to what I have written in regards to low-active builtin LEDs above, there is another problem with the sketch: Every JLed effect has a duration/period. Using `On()` or `Off()` configures internally an effect with period of 1ms. So when you call the `Off()`  after the `delay(2000)` the JLed effect is already finished and nothing will happen. Solution:

* Instead of calling `led.Off().Update()`, call `led.Stop()` to stop any effects and turn the LED off. This is the general approach to stop an effect and turn a LED off.
* Alternatively reset and reconfigure the `led` object by calling `Reset()`: `led.Reset().Off().Update()`. This approach can be used to assign a different effect to a JLed object.

Perhaps  `Reset()` should be always called when a new effect is assigned, I will think of making a change here.

iZhangHui, 2019-06-20:
@jandelgado Thank you very much for your quickly response.

I have tested on Arduino UNO and ESP32. They have the same results. The builtin LED is high-active.

After many experiments, I also explored the solution, which is same with your solution 2.

But I think it is not simple enough.  Looking forward to your optimization. :)

jandelgado, 2019-06-20:
Yes - I think so too. Will fix it the next days, stay tuned.

jandelgado, 2019-06-20:
New Release 4.2.0 addresses issue. Looking forward getting your feedback

iZhangHui, 2019-06-21:
Verified release 4.2.0, it works as expected. Great job!  :thumbsup:

closed 2019-06-21 by iZhangHui