jandelgado/jled #27: HardFault on SAMD21

Redferne, 2019-08-20:
Great library!, works splendidly on STM32 and ESPs. Tried running jled on a SAMD21 ([PlatformIO](https://github.com/platformio/platform-atmelsam) [Framework](https://github.com/platformio/platformio-pkg-framework-arduinosam) But is crashes with HardFault here: https://github.com/jandelgado/jled/blob/0de3a09cdef1c9d1320062f6d20aabfe884e6970/src/jled_base.h#L249
All v4 versions have this issue, v3 works fine. Did not have time to attach the debugger. Thanks!

jandelgado, 2019-08-22:
Since I have no SAMD21, could you please try the following:
In file jled.h change the `#ifdef ESP8266` to `#if 1` and test if this fixes the problem.

https://github.com/jandelgado/jled/blob/0de3a09cdef1c9d1320062f6d20aabfe884e6970/src/jled_base.h#L422-L427

Redferne, 2019-08-22:
Yes. I already tried that. I was thinking the same thing 😂 But unfortunately it did not help 😭

jandelgado, 2019-08-24:
I just ordered a SAMD21 dev board ;) Could you please provide some demo code so I can reproduce the problem?

jandelgado, 2019-08-24:
In branch fix_samd21_hard_fault I slightly changed memory alignment code, perphaps this does the trick. If not I've to wait for my SAMD21 dev board to further investigate.

jandelgado, 2019-08-30:
I successfully tested latest version 4.2.1 with a Sparkfun SAMD21 dev board. I can also confirm that 4.2.0 did not work.

PR #29 (2019-08-30): Fix samd21 hard fault
Fixes #27 
Problem of misaligned memory should now also be fixed for other platforms, since c++ keywords alignas/alignof are used.
closed 2019-08-30 by jandelgado