jandelgado/jled #30: Question: What should I expect from following code?

rahuldeo2047, 2019-09-15:
Hey, thank you for creating the lib. It is fun to play with it.

I have some doubts.

```
JLed(LED).DelayBefore(150).Blink( 4650, 50).Repeat(20).DelayAfter(150).Breathe(5000).Forever();

//Why only breath is working?

```
and
```
JLed(LED).DelayBefore(150).Breathe(5000).Repeat(20).DelayAfter(150).Blink( 4650, 50).Forever();

//Why only blink is working?
```

I expected delay 150ms, blink (4650,50), repeat 20 times, delay for 150ms and then breath for 5000ms and repeat these all forever.

Anything wrong I am doing?

jandelgado, 2019-09-16:
Hi, per JLed Object, only a _single_ effect can be active per instance. In your examples, this would be the last effect you set. To play multiple "effects", have a look at the `JLedSequence` class and the [examples](https://github.com/jandelgado/jled/blob/master/examples/sequence/sequence.ino).

rahuldeo2047, 2019-09-18:
Oh complete my bad.
I have gone through the example but sone how missed the point.

closed 2019-09-18 by rahuldeo2047