jandelgado/jled #31: Is it possible to append a sequence?

sslupsky, 2019-09-16:
Regarding the sequence class, is it possible to dynamically append to a sequence?  For example, I would like to queue a sequence with some number of effects and then sometime later before the effects are complete, I would like to queue another effect to the end of the sequence.  Can this be done with the existing library?

jandelgado, 2019-09-20:
Hi, this is currently not possible, since the array of `JLed` objects passed to the `JLedSequence` constructor is not managed by the `JLedSequence` object, but the calling client (since I want to keep memory management out of the lib) and currently the array can't be changed after construction. 

Easiest way to achieve what you want to do with the current lib would be to construct a second `JLedSequence` object with a new or modified `JLed` object array and switch to this one.

sslupsky, 2019-09-21:
That is what I thought.  Thanks for the feedback.  What I am trying to accomplish is to queue an effect.  I thought sequence might help this.  The approach I have taken is to use a queue object.  At first, i thought I could queue the JLed object but realized that is likely not going to work because the constructor initializes the time_start_ and last_update_time_ et al.  So I created a struct for the blink effect parameters and pull the struct from the queue and do an assignment to a JLed blink object.  There is an issue with this approach however.  You cannot change the effect from say Blink to Breathe.  I tried to create an uninitialized JLed object and then dynamically assign the effect but this would not compile.

jandelgado, 2019-09-23:
I quickly assembled an example (could not test it, because I have no micro controller at hand right now). The example randomly "plays" effects, it's a similar use case to yours, hope that helps:
```c++
// Copyright 2019 by Jan Delgado. All rights reserved.
// https://github.com/jandelgado/jled
#include <jled.h>

// Endless playing random JLed sequence
class RandomJLedSeq {
    private:
        uint8_t pin_;
        JLed *led_ = nullptr;

        // return a new JLed object where (most) parameters are set randomly.
        JLed* NewRandomJLedSeq() const {
            auto led = new JLed(pin_);
            switch(random(0,5)) {
                case 0: led->Blink(random(100,2000), random(100,2000)); break;
                case 1: led->Breathe(random(100,2000)); break;
                case 2: led->FadeOn(random(100,2000)); break;
                case 3: led->FadeOff(random(100,2000)); break;
                case 4: led->Candle(); break;
            }
            led->Repeat(random(2,11));
            led->DelayBefore(1000);
            return led;
        };

    public:
        RandomJLedSeq(uint8_t pin) : pin_(pin) {}

        virtual ~RandomJLedSeq() { delete led_; }

        RandomJLedSeq(const RandomJLedSeq& led) { *this = led;}

        RandomJLedSeq& operator = (const RandomJLedSeq&led) {
            if (this != &led) {
                pin_ = led.pin_;
                led_ = led.led_ ? new JLed(*led.led_) : nullptr;
            }
            return *this;
        }

        void Update() {
            if (!led_) {
                led_ = NewRandomJLedSeq();
            }
            auto running = led_->Update();
            if (!running) {
                delete led_;
                led_ = nullptr;
            }
        }
};

auto led = RandomJLedSeq(LED_BUILTIN);

void setup() {
}

void loop() {
    led.Update();
}
```

sslupsky, 2019-09-24:
Thanks for the feedback.  I have had some success with using a queue.  Along the way, I think I have observed that this library has a problem when used with interrupts.  As mentioned in the other thread, I use my own timer for keeping track of time when I sleep.  I use the SAMD "periodic event" timer.  I have a method "blink" that I use to queue an LED blink.  The periodic event calls Update() and checks to see if there is a pending blink in the queue.  If there is, it pops the blink from the queue and assigns the effect parameters to the JLed object.  I have determined that the flags_ variable is optimized by the compiler and doesn't update properly.  That is, when I check the status using IsRunning() in my loop(), it is always true.  If I add the "volatile" keyword to the declaration then IsRunning() returns the correct state.  

`    volatile uint8_t flags_ = 0;
`

I have not considered yet if any of the other variables need to be declared volatile as well.

jandelgado, 2019-09-26:
JLed is not thread safe. If you run it in a multi threaded environment, you have to take care yourself. Could you post some example code so I better understand the setup?

sslupsky, 2019-09-26:
Here is an example:

```
//  JLed
//  create queue for LED
Queue	ledBlink_q(sizeof(ledBlink_t), 10, FIFO);
JLed statusLED = JLed(LED_BUILTIN).Blink(1000, 1000).Repeat(1);

void blink( int numBlink = 1, unsigned int speed = BLINK_DURATION_SHORT, int level = BLINK_LEVEL_VERBOSE, int color = 1, bool wait = false ) {
  if ( level <= BLINK_LOG_LEVEL ) {
    ledBlink_t led;
    led.on = speed;
    if ( speed < (RTC_PERIODIC_EVENT_PERIOD * 2) ) {
      led.on = RTC_PERIODIC_EVENT_PERIOD * 2;
      led.off = RTC_PERIODIC_EVENT_PERIOD * 2;
    } else {
      led.on = speed;
      led.off = speed;
    }
    led.num = numBlink;
    led.wait = wait;
    noInterrupts();
    ledBlink_q.push(&led);
    interrupts();
    if ( led.wait ) {
      while ( !ledBlink_q.isEmpty() || statusLED.IsRunning() ) ;
    }
  }
}

void rtcPeriodicEvent() {
// EVSYS periodic event interrupt callback
  rtcTickMS += RTC_PERIODIC_EVENT_PERIOD;
  //  millis() in jledhal needs to be updated for this to work during sleep.
  if ( !statusLED.Update() ) {
    if ( !ledBlink_q.isEmpty() ) {
      ledBlink_t led;
      ledBlink_q.pop(&led);
      statusLED = JLed(LED_BUILTIN).Blink(led.on, led.off).Repeat(led.num);
    }
  }
}

setup() {
  blink(5, BLINK_DURATION_LONG, BLINK_LEVEL_NOTICE, 1, false);
}

loop() {
}
```

I updated millis() in arduino_hal.h to return the value of rtcTickMS.

The rtcPeriodicEvent() handler processes the rtc periodic event interrupt.  At the moment, this interrupt fires every 125 ms.  The handler calls the JLed Update() method.  This is done so that the LED timing is not affected by the length of time it takes to execute the code in the loop() which can vary depending on what it is doing at any particular time.   When there is no running JLed, the handler pops the parameters off the queue and starts a new JLed.

The blink() method pushes the parameters for the blink onto the queue.

Without the volatile declaration, IsRunning() always returns true. 

The queue must also be interrupt / thread safe.  For the moment, I brute force it and turn off the interrupts when pushing onto he queue.  I have found another library for this purpose and will use that shortly.

closed 2021-02-24 by github-actions[bot]