jandelgado/jled #32: Question:  Can millis() be replaced with another time method

sslupsky, 2019-09-20:
I would like to use jled in a low power application.  I have set up the periodic event system (on the SAMD21) to wake the mcu every 125ms.  I noticed that the jledhal uses the millis() function to determine the time.  The problem with millis() in a low power application is that this timer stops while the mcu is sleeping.  I keep track of time using another custom method that tracks elapsed time.

Is it possible to replace the call to millis() with a call to a custom time method?  I think the following line needs to change in arduino_hal.h:

`    uint32_t millis() const { return ::millis(); }
`

Can I simply change it as follows:

`    uint32_t millis() const { return myClass::newmillis(); }
`

Is there an way to do this without changing the jled library?

jandelgado, 2019-09-21:
Yes, you can define you own custom HAL class, which controls access to the GPIO and the timer. I have assembled an example, which will be included in the examples of JLed. 
Is this what you are looking for?

```c++
// JLed custom HAL example. 
// Copyright 2019 by Jan Delgado. All rights reserved.
// https://github.com/jandelgado/jled

// we include jled_base.h instead of "jled.h" since we define our own JLed
// class using our custom HAL.
#include <jled_base.h>

// a custom HAL for the Arduino, inverting output and ticking with half
// the speed.
class CustomHal : jled::JLedHal {
    private:
        template <typename T, typename B>
            friend class jled::TJLed;
        CustomHal() {}

    public:
        explicit CustomHal(uint8_t pin) noexcept : pin_(pin) {}

        void analogWrite(uint8_t val) const {
            // some platforms, e.g. STM need lazy initialization
            if (!setup_) {
                ::pinMode(pin_, OUTPUT);
                setup_ = true;
            }
            ::analogWrite(pin_, 255-val);
        }

        uint32_t millis() const { return ::millis() >> 1; }

    private:
        mutable bool setup_ = false;
        uint8_t pin_;
};

class JLed : public jled::TJLed<CustomHal, JLed> {
    using jled::TJLed<CustomHal, JLed>::TJLed;
};

// uses above defined CustomHal
auto led = JLed(LED_BUILTIN).Blink(1000, 1000).Repeat(5);

void setup() {
}

void loop() {
    led.Update();
}
```

sslupsky, 2019-09-24:
Yes, I modified the arduino_hal.h to change the millis().  I was wondering if this could be part of the API where you call a method to override millis() with your own function?

jandelgado, 2019-09-26:
The official, non-invasive, way would be to implement a custom HAL as shown above. This way, no modification of the lib is necessary.

sslupsky, 2019-09-26:
Wouldn't you still need to change the reference to arduino_hal.h in the code snippet below from jled.h to the custom HAL ?

```
#ifdef ESP32
#include "esp32_hal.h"  // NOLINT
namespace jled {using JLedHalType = Esp32Hal;}
#elif ESP8266
#include "esp8266_hal.h"  // NOLINT
namespace jled {using JLedHalType = Esp8266Hal;}
#else
#include "arduino_hal.h"  // NOLINT
namespace jled {using JLedHalType = ArduinoHal;}
#endif
```

jandelgado, 2019-09-28:
No, because in the custom HAL example above, I included `jled_base.h` instead of `jled.h` and defined my own `JLed` class. This is basically what is usually done in `jled.h`, for convenience reasons.

jandelgado, 2019-11-16:
Hi, can we close the issue? The problem can be solved as described. In the meantime, I added an Custom-HAL example to the [examples](https://github.com/jandelgado/jled/blob/master/examples/custom_hal/custom_hal.ino) section.

br 
Jan

closed 2020-01-17 by jandelgado