jandelgado/jled #34: Change time of efect in loop function

petione, 2019-09-24:
How I can change time of effect in loop function or add variable as time?

#include <Arduino.h>
#include<jled.h>
// blink internal LED every second; 1 second on, 0.5 second off.
auto led = JLed(LED_BUILTIN).Blink(100, 500).Forever();
void setup() {}
 void loop()
{
   led = JLed(LED_BUILTIN).Blink(1000, 500).Forever(); //this not work
  led.Update();
}

jandelgado, 2019-09-26:
This does not work because you are re-assigning the `led` object again and again in the `loop()` function. You need some kind of a trigger and re-assign only once, e.g. (not tested):
```c++
...
auto changed = false;
...
void loop() {
     static auto time_start = millis();
     if (mills() - time_start > 5000 && !changed) {
         changed = true;
         led = JLed(LED_BUILTIN).Blink(1000, 500).Forever(); 
    }   
    led.Update();
}
```

petione, 2019-09-26:
I'm understand. Thanks for reply.

closed 2019-09-26 by petione