jandelgado/jled #36: auto led = JLed(ledPin).Blink(1, 1000).Forever(); doesn't does not blink regularly.

avandalen, 2019-11-11:
What can you program well! I am overwhelmed with this library...
just:
auto led = JLed(ledPin).Blink(1, 1000).Forever(); doesn't does not blink regularly.
auto led = JLed(ledPin).Blink(2, 1000).Forever(); is okey

jandelgado, 2019-11-15:
`Blink(1,1000)` means to turn the LED on for 1ms and to turn it off for 1000ms = 1 second afterwards. I did not test it, but I doubt that this 1ms "flash" will be visible to the human eye. 

What do you exactly expect?

avandalen, 2019-11-16:
It is funny, Blink(1, 1000) is still visable, look what happens...
To save power, I use 2ms in my application.

From: jandelgado 
Sent: Friday, November 15, 2019 9:55 AM
To: jandelgado/jled 
Cc: Albert van Dalen ; Author 
Subject: Re: [jandelgado/jled] auto led = JLed(ledPin).Blink(1, 1000).Forever(); doesn't does not blink regularly. (#36)

Blink(1,1000) means to turn the LED on for 1ms and to turn it off for 1000ms = 1 second afterwards. I did not test it, but I doubt that this 1ms "flash" will be visible to the human eye.

What do you exactly expect?

jandelgado, 2019-11-16:
So I was curious to see if there might be a bug or some unexpected behaviour. I connected a LED to an Arduino nano and checked with an Oscilloscope if output is as expected. Luckily it is:

* `Blink(1,1000)`
[image]

* `Blink(2,1000)`
[image]

* Both effects have a period of 1000ms:
[image]

Everythin works as expected, I do not see an issue. Can we close the issue?
br Jan

avandalen, 2019-11-17:
In my case the LED was not blinking regulary with 1ms (visually) , but I have not checked it with my oscilloscope, i will check that, you can close the isue
thank you very much

From: jandelgado 
Sent: Saturday, November 16, 2019 4:12 PM
To: jandelgado/jled 
Cc: Albert van Dalen ; Author 
Subject: Re: [jandelgado/jled] auto led = JLed(ledPin).Blink(1, 1000).Forever(); doesn't does not blink regularly. (#36)

So I was curious to see if there might be a bug or some unexpected behaviour. I connected a LED to an Arduino nano and checked with an Oscilloscope if output is as expected. Luckily it is:

  a.. Blink(1,1000)

  b.. Blink(2,1000)

  c.. Both effects have a period of 1000ms:

Everythin works as expected, I do not see an issue. Can we close the issue?
br Jan

closed 2019-11-17 by jandelgado