jandelgado/jled #37: Adding diagnostic blinking LED error code

avandalen, 2019-11-14:
I made a heartbeat LED library, but your library is better. In my library there is normally a continuous heartbeat but in the case of an error the LED indicates the error code. Do you want to integrate this? See here:
https://github.com/avandalen/avdweb_HeartBeat
https://www.avdweb.nl/arduino/hardware-interfacing/heartbeat

jandelgado, 2019-11-18:
Good idea. It's already possible:

```c++
#include <jled.h>

class Heartbeat {
    JLed led;
    bool inError = false;

public:
    Heartbeat(uint8_t pin) : led(JLed(pin)) {
        led = JLed(pin).Breathe(1000).Forever();
    }
    bool Update() {return led.Update();}
    void Error(uint8_t code) {
        if (inError) return;
        inError = true;
        led.Blink(500,100).Repeat(code);
    }
};

// attach heartbeat LED to PIN D9
auto heartbeat = Heartbeat(9);

void setup() { }

void loop() {
    // condition turns true after ~ 3s
    const auto condition = (millis()/100 == 30);

    if (condition) {
        // Enter error mode
        heartbeat.Error(3);
    }
    heartbeat.Update();
}
```
So I rather wouldn't add a new API for this feature but add a new example.

jandelgado, 2019-11-18:
Tracked here: https://github.com/jandelgado/jled/projects/1#card-29397135

jandelgado, 2019-11-18:
I just noticed that it would be great if the `Blink()` error would repeat after a pause, instead of firing just once. Will find a solution...

closed 2021-02-24 by github-actions[bot]