jandelgado/jled #39: How to combine FadeOn(), On(), and FadeOff()

RossAWaddell, 2019-11-15:
Fantastic library! I've implemented a number of your functions already in my project but I'm stumped on how to do this: I need to simulate an old, incandescent blinking Christmas tree light which involves:

1. Fade up over a period of 100ms
2. Stay on for a period of 1500ms
3. Fade off over a period of 100ms
4. Stay off for period of 500ms
5. Repeat

(This is different than your Breathe() function which fades up for half the set period and then immediately fades off for the other half)

Is this possible with the Sequence() function? I looked at the example but saw there was a "delay(1)" in the loop which I need to avoid.

jandelgado, 2019-11-16:
Yes, the  `JLedSequence` could be used for that. The `delay(1)` in the example is not needed, I just added it to simulate some other work done in the main `loop()`. However `JLedSequence` does not yet support the `Repeat` step. I'll add an backlog item to add the `Repeat` feature to the `JLedSequence` class.

RossAWaddell, 2019-11-16:
Thanks very much, jandelgado. I look forward to that enhancement.

jandelgado, 2019-11-18:
Will be tracked here: https://github.com/jandelgado/jled/projects/1

closed 2019-11-18 by jandelgado