jandelgado/jled #40: Simple use case: How to blink only one time and stop

fedevo, 2019-11-17:
Hi jandelgado,
really a great job !
Unfortunately I have a very simple use case, and I'm not sure how to use your lib.
Based on a trigger, I simply need a unique non-blocking blink (on for 100ms then off) until the next trigger.
For the moment I use your lib like this:
```
led.Blink(100, 0).Repeat(1).Update(); -- NOK, led stay on after the first trigger
led.Blink(100, 1).Repeat(1).Update(); -- OK, on for 100ms then off
```
is there a simpler way to achieve this ?
If not, I would suggest to add an interface that could be used like this:
```
led.On(100).Update();
```
What do you think ?
Regards
Vince

jandelgado, 2019-11-18:
Nice edge case - never thought about this one ;)

Some notes:
* As you noted, `Blink(100,0)` will not turn the LED off since the off-duration is 0, so you have to provide a non-zero second parameter (works as designed)
* `Repeat(1)` can be omitted, since it is the default

Here is how I would probably do it:
```c++
#include <jled.h>

// Create JLed object in an inital stopped stated, i.e. IsRunning() will return false.
auto led = JLed(13).Stop();

void setup() {}

void loop() {
    // sample condition turns true after approx 3 seconds
    const auto condition = (millis()/100 == 30);

    if ( condition && !led.IsRunning())  {
        // blink once for 500ms
        led.Blink(500, 1);
    }
    led.Update();
}
```

I'll think of providing a neater API for the use case. But for the moment the problem can be solved as shown above.

fedevo, 2019-11-18:
Thanks for your feedback and clarifications, your proposed solution cover perfectly my need.
It was my first usage of your lib and I think I will use it more extensively in a near futur, will let you know if I find other edge cases ;-)

closed 2019-11-18 by fedevo

jandelgado, 2019-11-20:
Excellent! Always like to get feedback and suggestions.