jandelgado/jled #43: Different brightness level in effects

Kot-Zakhar, 2019-12-07:
It would be very cool for you to implement option of setting max brightness of effect.
Currently trying to use your library with RGB led and missing this simple feature very much...

Kot-Zakhar, 2019-12-07:
UPD:
Found the same question in solved ones. Seems like a good feature to implement :)
BTW: Very cool library. Love this fluent language-like syntax. Approved 👍

jandelgado, 2020-01-17:
Yes, this would be a nice feature. I'll put a story it in the backlog.

vidojev, 2020-01-24:
**Great library**. I do enjoy making led run, also I use blink for audio buzzer. Strongly support option to set max level in effects!

KurTell, 2020-02-07:
Great library, works like a shot! Thanks for sharing.
I also vote for support for changing the brightness. As a workaround I divided the return value of fadeon_func in jled_base.cpp by 4 or 8.

jandelgado, 2020-02-13:
I just made a few tests with this new feature. Without increasing the memory usage of a JLed object, we could provide 32 Levels of brightness. Would that be enough for your use cases?

vidojev, 2020-02-14:
Anything more than what exist today is good. For me. Thank you.

KurTell, 2020-02-15:
32 levels should be all I need. Btw. tempering with the fadeon_func did not work. Thanks for your quick response!

jandelgado, 2020-02-27:
Feature is implemented in Release 4.5.0 (`MaxBrightness() method)

sixtyfive, 2020-06-02:
Any chance we might also get a `MinBrightness` to go along with it? :-)

Kot-Zakhar, 2020-06-13:
@jandelgado Could you tell plz, when should I expect this release (4.5.0) to appear in platformio registry?
Is there smth need to be done to see it there? Or should we just wait for it?

jandelgado, 2020-06-13:
now it is ;-) seems that I forgot to bump the version in `library.properties` - thanks for the hint.

Note that besides of using git tags, you can also use commit-hashes in `platformio.ini`, e.g.

```                                                                               
 lib_deps = https://github.com/jandelgado/jled#b7bd5d8fd3f68582ded83a7f93e563fc371f0845
```
by the way, could ou then please close the issue.

br jan

closed 2020-07-18 by jandelgado