jandelgado/jled #44: Sequences and single setups on same PIN

vidojev, 2020-01-24:
When sequence and single setup are "working" on the same pin (PWM 44/45 on Mega) there are some misfire on other Mega pins. Code goes wild. When pins in sequence are different than in single setup there are no issues. Very strange behaviour.

jandelgado, 2020-01-24:
Intersting ;) Could you please provide a full sketch so I safely reproduce it? I have a Mega somewhere around here. Thanks!

vidojev, 2020-01-24:
Of course, but it is not done yet and most of its comments are in Serbian but you will find what you are looking for I’m sure… It is a code for a light switch for my daughter and living room that is having an option to select individual and group of lights, dim it (that is not in the code yet), etc... Thank you for reply very much!Vidoje. /*  Light_Switch_Universal  11.01.2020.@23:24*/ //biblioteke#include "jled.h"               //https://github.com/jandelgado/jled#include "EButton.h"            //https://github.com/jonnieZG/EButton //globalne konstante#define delay_step 15#define delay_stop 1000#define selektor_timeout 3000#define ljuljanje 18#define ljulj_puta 4 //Mora biti paran broj ljuljanja!  //globalne konstante za JLed#define broj_flasheva 3#define flash_delay 65#define off_led_level 25 //globalne konstante za EButton#define DebounceTime 75#define ClickTime 350#define LongPressTime 900 //Lara_setup#define L_prekidac 39#define Larin_prekidac_LED 44#define L_selektor 31#define Larin_selektor_LED 45#define Larin_buzzer 33 //Dnevna setup#define D_prekidac 46#define Dnevna_prekidac_LED#define D_selektor 48#define Dnevna_selektor_LED#define Dnevna_buzzer 33 //init JLedauto Lara_buzzer = JLed(Larin_buzzer).Blink(3, 15).Repeat(2);auto Lara_prekidac_LED = JLed(Larin_prekidac_LED).Blink(flash_delay, 40).Repeat(broj_flasheva);auto Lara_selektor_LED = JLed(Larin_selektor_LED).Blink(flash_delay, 40).Repeat(broj_flasheva);auto Lara_prekidac_LED_long = JLed(Larin_prekidac_LED).Blink(flash_delay, 80).Forever();auto Lara_selektor_LED_timeout = JLed(Larin_selektor_LED).DelayBefore(selektor_timeout).Breathe(300).Repeat(2); /*  JLed test_run[] = {  JLed(Larin_prekidac_LED).Breathe(1200).Forever(),  JLed(Larin_selektor_LED).DelayAfter(600).Breathe(1200).Forever()  };*/   JLed test_run[] = {  JLed(Larin_prekidac_LED).Set(10).Forever(),  JLed(Larin_selektor_LED).Set(10).Forever()  };  JLedSequence abc(JLedSequence::eMode::PARALLEL, test_run);  //init EButtonEButton Lara_prekidac(L_prekidac);EButton Lara_selektor(L_selektor); int funkcija[] = {  0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29,  30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 48, 49, 50, 51, 52, 54, 55, 56, 57, 59, 60,  61, 63, 64, 66, 67, 69, 70, 72, 73, 75, 76, 78, 80, 81, 83, 85, 87, 89, 90, 92, 94, 96, 98, 100, 102, 104, 106,  108, 110, 112, 115, 117, 119, 121, 124, 126, 128, 131, 133, 136, 138, 141, 143, 146, 149, 151, 154, 157, 160, 162,  165, 168, 171, 174, 177, 180, 183, 187, 190, 193, 196, 200, 203, 206, 210, 213, 217, 220, 224, 228, 231, 235, 239,  243, 247, 251, 255}; struct a //svetla{ const byte in_pin;  const byte count;  const byte out_pin;  int pravac;  bool stanje;  unsigned int nivo;}; struct b //grupe{  const byte in_pin;  const byte selekcija;  const byte count;  const byte out_pin_id;};  struct c //izbor(selekcija){  const byte pin_p;  const byte pin_s;  const byte total_selekcija;  byte izbor_grupe;  const byte long_slc_izbor;  const byte long_slc_nivo;}; a svetla[] = {  {D_prekidac, 4, 2, 1, 1, 67},  {0, 3, 3, -1, 1, 99},  {0, 2, 4, 1, 0, 100},  {0, 1, 5, -1, 1, 123},  {L_prekidac, 2, 8, -1, 0, 133},  {0, 1, 9, -1, 0, 133}}; b grupe[] = {  {L_prekidac, 3, 2, 4},  {0, 0, 1, 5},  {L_prekidac, 2, 1, 4},  {L_prekidac, 1, 1, 5},  {D_prekidac, 7, 4, 0},  {0, 0, 3, 1},  {0, 0, 2, 2},  {0, 0, 1, 3},  {D_prekidac, 6, 2, 0},  {0, 0, 1, 3},  {D_prekidac, 5, 2, 1},  {0, 0, 1, 2},  {D_prekidac, 4, 1, 0},  {D_prekidac, 3, 1, 1},  {D_prekidac, 2, 1, 2},  {D_prekidac, 1, 1, 3}}; c izbor[] = {  {L_prekidac, L_selektor, 3, 3, 3, 25},  {D_prekidac, D_selektor, 7, 7, 6, 100}}; const int broj_pozicija = (sizeof(funkcija) / sizeof(funkcija[0]) - 1);const byte broj_svetla = (sizeof(svetla) / sizeof(svetla[0]));const byte broj_grupa = (sizeof(grupe) / sizeof(grupe[0]));const byte broj_selekcija = (sizeof(izbor) / sizeof(izbor[0])); unsigned long currentMillis, previousMillis, stepMillis, staro_vreme;/********************************************************/ void double_click_selector(EButton & taster) {  Serial.println();  Serial.println("DB Selektor ");  int i, selekcija, pin_p, poz_grupe;  int pin = taster.getPin();   //nadji izbor za dbclik, nivo i pin prekidaca  for (i = 0; i < broj_selekcija; i++) {    if (izbor[i].pin_s == pin) {      selekcija = izbor[i].total_selekcija;      izbor[i].izbor_grupe = selekcija;      pin_p = izbor[i].pin_p;      break;    }  }   //nadji poziciju u tabeli  for (int j = 0; j < broj_grupa; j++) {    if (grupe[j].in_pin == pin_p && grupe[j].selekcija == selekcija) {      poz_grupe = j;      break;    }  }   //nadji koliko ima istovremenih paljenja/gasenja  int petlja = grupe[poz_grupe].count;   //promeni stanje na svim pinovima u grupi (100%/ON/pravac -1)  for (int k = 0; k < petlja; k++) {    int red = grupe[poz_grupe + k].out_pin_id;    svetla[red].pravac = -1;    svetla[red].stanje = 1;    svetla[red].nivo = broj_pozicija;  }   //signaliziraj  signalizacija(pin);   //izvrsi paljenje/gasenje  on_off(petlja, poz_grupe);   //zapamti kada si bio  staro_vreme = currentMillis;   //signaliziraj timeout  signalizacija_timeout(pin, true);}  void long_start_selector(EButton & taster) {  Serial.println();  Serial.println("Long Selektor ");  int i, selekcija, pin_p, poz_grupe, level_50;  int pin = taster.getPin();   //nadji izbor za long start, nivo i pin prekidaca  for (i = 0; i < broj_selekcija; i++) {    if (izbor[i].pin_s == pin) {      sel [... cut ...] prekidac.tick();  Lara_selektor.tick();    abc.Update();  Lara_buzzer.Update();  Lara_prekidac_LED.Update();  Lara_selektor_LED.Update();  Lara_selektor_LED_timeout.Update();}     Sent from Mail for Windows 10 From: jandelgadoSent: subota, 25. januar 2020. 00.03To: jandelgado/jledCc: vidojev; AuthorSubject: Re: [jandelgado/jled] Sequences and single setups on same PIN (#44) Intersting ;) Could you please provide a full sketch so I safely reproduce it? I have a Mega somewhere around here. Thanks!

vidojev, 2020-01-24:
> Intersting ;) Could you please provide a full sketch so can I safely reproduce it? I have a Mega somewhere around here. Thanks!
> Edit: I just found this thread: https://forum.arduino.cc/index.php?topic=202045.0 , could it be related to your problem?

I can't say, it may have issues with Mega timers, but only if they are same "delays". Maybe because their least common denominator is 1... if it is huge i.e. 10/11 than maybe timers are not "called" at the same time...

closed 2021-02-24 by github-actions[bot]