jandelgado/jled #46: Chain sequences together

3dluvr, 2020-02-12:
Hi,

I have a use case where we need to chain a number of sequences together into "emotions", for an art installation. The sequences in an "emotion" should play one after another, but only upon an external trigger. How do we go about that?

It appears there's no way to know whether the sequence is running, or has completed. Basically, a flag where its known if a specific sequence is playing, has already played out, might allow to chain run them one after another, in a specific (or random) order.

Thanks.

jandelgado, 2020-02-12:
If your are using the `JLedSequence` class for the sequence, you can use the return value of the update method:

https://github.com/jandelgado/jled/blob/7219c3a0f0afaa39becdb3b7fc502c978e892dcf/src/jled_base.h#L472

Which returns `true` as long as the sequence is running, otherwise `false`.

3dluvr, 2020-02-12:
Does that not also execute one cycle of the sequence as well?

Ideally, I'd like to be able to create a sequence in a disabled state. Even if I call .Update() nothing would happen because it is disabled.
Then if I change some flag (.enabled()) it would start upon next .Update() loop.

Also, knowing whether the sequence already ran or not would be awesome. That way I don't want it to run if it already executed, until I reset it. Another flag perhaps that says whether the sequence completed. .Update() only tells me whether it's currently running or not.

closed 2021-02-24 by github-actions[bot]