jandelgado/jled #51: Various questions

harueg, 2020-03-02:
Hi
Thanks for this great software !
What exactly is non-blocking, something similar to multitasking like with freertos ?
Would it be possible to change the start and end values ( brightness ) on these effects, especially breathe ? 
Do you know these flameless candles ?
I would like to build something similar, just bigger.
I would have 4 solenoids moving the flame and 5 channels for the LEDs.
I would use a ESP32 to have some kind of remote control : ON / OFF, Brightness, Speed.

Thanks and best regards

Harald

jandelgado, 2020-03-03:
Hi,

> What exactly is non-blocking, something similar to multitasking like with freertos ?

It basically means that your sketch/program using `JLed` will not be blocked while the effects play. This is achieved by calling the `Update` method on the `JLed` objects from inside your main loop, like e.g. in `void loop() { led.Update(); }`, instead of doing something like e.g. 
```c++
// blocking example
void loop() {
   analogWrite(LED_BUILTIN, 0);
   delay(500);
   analogWrite(LED_BUILTIN, 255);
   delay(500);
}
```
for a blinking LED, which will block your code in the `delay` calls. JLed does not use "multi-tasking" or timers, the client has to make sure that the `Update` method is called regulary.

See the [examples](https://github.com/jandelgado/jled/blob/master/) for details.

> Would it be possible to change the start and end values ( brightness ) on these effects, especially breathe ?

Yes, use the `MaxBrightness` method to [set the maximum brightness of a LED](https://github.com/jandelgado/jled#maximum-brightness-level). It allows to set the maximum level of an effect.

> Do you know these flameless candles ?

Yes, use the `Candle` effect to [simulate candles and fire flickering](https://github.com/jandelgado/jled#candle).

> I would like to build something similar, just bigger.
>
> I would have 4 solenoids moving the flame and 5 channels for the LEDs.
> I would use a ESP32 to have some kind of remote control : ON / OFF, Brightness, Speed.

Sounds like an interesting project, looking forward to see it in action ;)

harueg, 2020-03-03:
Hi jandelgado
Thanks for the quick response !
The candle effect is a bit to flickery for my application. I really do like the breathing.
I just need to change the parameters on the fly.
i.e. min. brightness, max. brightness, duration. These values should be ( in limited range ) random generated. The limiting range then is controlled by an external input, preferably over wifi or something similar. ( when time allows )
I added a nice candle effect, which for some unknown reason worked on an UNO.
But I wasn't able to compile it for the ESP32. No idea whether it was multitasking on the UNO.
The code for one LED is from line 75 - 144. It dims between the old and the new brightness value.
I think you do something similar, just with more math. As far as I could decipher from the sources.

Yes... action,  it has to be finished in 2 weeks.

Thanks again

Harald

[blink_analogread_Led_ino5.zip](https://github.com/jandelgado/jled/files/4283251/blink_analogread_Led_ino5.zip)

closed 2020-06-26 by jandelgado