jandelgado/jled #52: Change exisiting effect

anam-zahid, 2020-03-19:
Hi,

so i am using esp32 to light up 5 different leds in a JLed array.
i am using random function to initialize the delay to breathe function of these leds when they are declared.
Now what i want is to again change the breathing times of these leds in the void loop(). Any way to that. 
Q. Is there any way i can change the running breathe effect during runtime to again call random and change its duration.
Note: Kindly also show a code that shows how to change a running effect in the FAQ. 
`#include <jled.h>
#define led1 27
#define led2 26
#define led3 25
#define led4 33
#define led5 32
#define freq  5000
#define ledChannel  0
#define resolution  8
int rnum [5];
int ledno[] = {led1,led2,led3,led4,led5};
//int numbering[]= {led1,led2,led3,led4,led5};
//const size_t n = sizeof(numbering) / sizeof(numbering[0]);
//int randomdelays;
int x =0;
JLed leds[] = {
  JLed(ledno[0]).Breathe(rnum[0]=random(2000,8000)).DelayAfter(1000).Forever(),
  JLed(ledno[1]).Breathe(rnum[1]=random(2000,8000)).DelayAfter(1000).Forever(),
  JLed(ledno[2]).Breathe(rnum[2]=random(2000,8000)).DelayAfter(1000).Forever(),
  JLed(ledno[3]).Breathe(rnum[3]=random(2000,8000)).DelayAfter(1000).Forever(),
  JLed(ledno[4]).Breathe(rnum[4]=random(2000,8000)).DelayAfter(1000).Forever()
  };
  JLedSequence sequence(JLedSequence::eMode::PARALLEL, leds);

void setup() {
  // put your setup code here, to run once:
 Serial.begin(115200);
}

void loop() {
sequence.Update();
}

`

jandelgado, 2020-03-20:
Hi, you could try to change the JLed effects in the `leds[]` with something like:

```c++
void loop() {
  if (some_somecondition) {
    leds[0].Breathe(1000);
    leds[1].FadeOn(2000);
   // etc.
  }
sequence.Update();
}

```

anam-zahid, 2020-03-21:
in the example you have given, can breathe have random(2000,8000) as it's parameters?.
Since i want the duration of breathe to change after previous breathe effect has finished.
To make it easier for you:
1. Assume that for the first led, random assigned a value of 3000 to breathe.
2. In the void loop, i again want random to assign another value, when the previous breathe effect (the one in point 1) had finished.
3. And so the process of randomly assigned values to breathe continues.

Essentially what i want is that:
1.  The led breathes at a different rate every time the object is updated.
2. Minimum code is required. Hence, i am avoiding sequential commands. Which i would have to give to every led separately. 
Extra Bit (You can ignore it if you want) : Even if i did do that, there would still be a sequence of breaths (lets say 2 short breath and 1 long breath) and thus not random. 

Oh and kindly increase documentation:
1. Like showing how to change a running effect at run-time. By that i mean in void loop().
2.  Demonstrate the working of state and misc  functions through an example code.

And by the way, this library is a life-saver, i am currently making automatic stairs lights, which will illuminate the step on which the person is standing, there will also be a music mode where the led's will act as a vu meter.

closed 2021-02-24 by github-actions[bot]