jandelgado/jled #53: Compiler fails invalid conversion from int to PinName Nano Sense 33 BLE

Naphier, 2020-06-06:
Nano Sense 33 BLE 

Compiler fails:
`invalid conversion from 'int' to 'jled::MbedHal::PinType {aka PinName}' [-fpermissive]`

Code: 
`auto led = JLed(13).Blink(1000,500).Forever();`
Also fails with `LED_BUILTIN`

jandelgado, 2020-06-06:
Hi @Naphier,
which environment (e.g. Arduino IDE / PlatformIO) and settings (Board/Framework) did you use?

jandelgado, 2020-06-06:
I installed the `Arduino nRF528x mbed Core ` into the Arduino IDE and can confirm that it does not compile. However, the following workaround (`#undef __MBED__`) does compile. Could you please try while I further investigate the issue:

```c++
#undef __MBED__
#include <jled.h>

auto led = JLed(13).Blink(1000,500).Forever();

void setup() {  
}

void loop() {
  led.Update();
}
```

Naphier, 2020-06-06:
Thanks. Just got the Nano Sense 33 and prefer this library over others (so many juicy options). 
Using Arduino IDE 1.8.18  (don't think that matters)
and  Arduino Nano 33 BLE (nRF528x - MbedOS)
Your workaround allows it to work, but I worry undef'ing __MBED__ will cascade to other things.

jandelgado, 2020-06-06:
Excellent - I'll provide a fix that does not make the `#undef` necessary. Thanks for testing

jandelgado, 2020-06-26:
Please check #54 for a fixed version - does it work for you?

closed 2020-07-18 by jandelgado

Naphier, 2020-07-21:
Solid. Thanks for the quick turnaround. Apologies that I couldn't test it sooner for you.

commit 0c656fbc (2020-07-18): add support for nano33ble using mbed arduino core (#54) 

The Arduino core for the nano33ble uses mbed and wraps it in the Arduino
core API. Therefore __MBED__ is defined, which makes JLed think it runs
in an Mbed environment. But in this case, JLed needs to use the Arduino
API. The problem is solved by checking also for ARDUINO_API_VERSION:
  #if defined(__MBED__) && !defined(ARDUINO_API_VERSION)
  -> running in mbed environment
in jled.h.

See
 * https://github.com/arduino/ArduinoCore-API
 * https://github.com/arduino/ArduinoCore-nRF528x-mbedos

Fixes #53