jandelgado/jled #56: Not able to turn blinking on after stopping it.

wesolykapselek, 2020-08-16:
Hi.

I'm trying to make a contraption where internal status will be passed to user with 3 LEDs which have to blink. If I turn off the LED I can't turn it on once again. Any suggestions? Device: Wemos D1 Mini.

https://pastebin.com/Ts89R1wf

jandelgado, 2020-08-16:
Hi,                                                                            
                                                                               
`On` and `Off` do not start and stop the current effect, they are effects         
themselves: if you call `Off` on a JLed object, then you replace the current   
effect (e.g. `Blink`) with a constant light effect with brightness 0 and period
1.                                                                                
                                                                                  
The same goes when you call `On`: you replace the current effect with a constant
light effect with brightness 255 and period 1.                                    
                                                                                  
So after calling `On` or `Off`, your `Blink` effect is gone and I suppose that 
is what you see.                                                                  
                                                                                 
Try to call `Blink(25,975)` or someting else again instead of `On` and see if  
this works.                                                                       
                                                                                  
See the documentation for the details: https://github.com/jandelgado/jled#static-on-and-off
                                                                               
br jan

wesolykapselek, 2020-08-16:
This is very kind of you that you responding on Sunday.

Tried two things like:
Assigning new object to array with construction like:
leds[redLEDindex] = JLed(redBlinkPin).Blink(500, 500).Forever();

Or like you suggesting call Blink or existing array element:
leds[redLEDindex].Blink(500, 500).Forever();

Both solutions ends with constant, full brightness light of LED.

jandelgado, 2020-08-18:
Look at this simple example I just tested on an ESP32. Initially it creates 3 blinking LEDs. After 2 seconds, the effects begin to change.

```c++
#include <jled.h>

JLed leds[] = {
    JLed(21).Blink(50, 100).Forever(),
    JLed(22).Blink(50, 100).DelayBefore(50).Forever(),
    JLed(23).Blink(50, 100).DelayBefore(100).Forever(),
};

JLedSequence sequence(JLedSequence::eMode::PARALLEL, leds);

void setup() { }

void loop() {
    static int state = 0;
    static int count = 0;

    sequence.Update();

    count++;

    if (count > 2000) {
        count = 0;

        switch(state) {
            case 0:
                leds[0].Off();
                leds[1].Off();
                leds[2].Blink(250,750).Forever();
                state = 1;
                break;
            case 1:
                leds[0].Off();
                leds[1].Breathe(500).Forever();
                leds[2].Off();
                state = 2;
                break;
            case 2:
                leds[0].FadeOff(500).Forever();
                leds[1].Off();
                leds[2].Off();
                state = 0;
                break;
        }

    }
    
    delay(1);
}
```

Looking again at your code the problem might be in your `loop` function, where you change the state of the LEDs over and over again depending on the value of `currentFlagStatus`.Shouldn't the LEDs only be changes when the value of `currentFlagStatus` changes?

wesolykapselek, 2020-08-18:
Damn! You are probably correct! I'm assigning the new object over and over and it can't really kick in!

wesolykapselek, 2020-08-18:
Your suggestion do the trick! Thank you very much!

closed 2020-08-18 by wesolykapselek