jandelgado/jled #59: Brightness Logic Inverted

ElectronicsBoi, 2020-09-19:
So i am using my esp32 wemos lolin32 board, for some reason the brightness logic is inverted. Set(255) cause LED to turn Off and Set( 0) causes LED to turn on. The Brightness scale is 255 to 0 , with 255 being the max brightness and 0 being the lowest. The max brightness function also doesn't have any change in the brightness of the led irregardless of the led effect being used. I think it may be a problem with this board or the code isn't best suited for some/all of the esp32 boards?

ElectronicsBoi, 2020-09-19:
I cant determine whether the breathe function looks fine or not, maybe because it involves both fade in and fade out, so it is difficult to determine which comes first. But one thing i did notice that it stays bright more than it stays dark. I don't know if it is the function's general behavior or due to the values being inverted in the waveform.

jandelgado, 2020-09-20:
I‘m not sure of I understood what your exact Problem is. But for inverted Outputs as described by you, you should try to use LowActive()

https://github.com/jandelgado/jled#low-active-for-inverted-output

br Jan

ElectronicsBoi, 2020-09-20:
Low active is just a fix for my board. I would appreciate it if the library is adapted somehow. Set(255) turns led off, fadein functions actually behaves as a fade off and fade off as a fade on. Max brightness cannot be used for some reason. I think if you have a lolin32 board, you should try your code on it. 

P.S The code works fine on my esp32 devkit board

ElectronicsBoi, 2020-09-20:
Yep, that was the problem, my on-board led was pulled low.

closed 2020-09-20 by ElectronicsBoi