jandelgado/jled #60: FadeOff leaves LED glimming

marmil, 2020-10-19:
When the FadeOff function reaches the last period step, the LED still glows as it is not fully switched off. There should be an implemented auto-off after end of fading time.

jandelgado, 2020-10-19:
Hi, 

could you please provide  complete, minimal example C++ code and an schematic of the circuit you used so I can check the behaviour?

Which version of JLed are you using?

marmil, 2020-10-20:
Hi,
thanks for your quick response.
I updated my original post with more information :)

jandelgado, 2020-10-20:
I did some experiments and it seems like we hit this ESP32 problem: https://github.com/espressif/arduino-esp32/issues/689

The LED stays dim on in this simple example:

```
#include <jled.h>       
                                                       
auto led = JLed(22).On().LowActive();                                         
void setup() { }                                                                              

void loop() {                                                                  
  led.Update();                                                                
}              
```

I'll provide a workaround in the next days

jandelgado, 2020-10-20:
See #61 for a fix - does it work for you?

jandelgado, 2020-10-23:
@marmil does the fix solve the problem?

marmil, 2020-10-24:
@jandelgado Works. Thank you 👍

jandelgado, 2020-10-24:
fix released with 4.5.2

closed 2020-10-24 by jandelgado