jandelgado/jled #62: ESP32 Unable to define channel

Johnymalina, 2020-11-18:
Hi,

i have problem with assign pins and channels using Esp32Hal(pin, chan). I get a message that Esp32Hal is not defined. I tried to include also esp32-hal.h library but without effect. Using only pin number works everythig fine but i would like to have more control of channels. I tried to explore what could cause this issue but the code is too complex for me. 

Thanks for your help.

Johnymalina, 2020-11-18:
I have managed to find a solution adding jled:: before jled helped. Is this expected behaviour? Maybe update to readme is in order:
Final code that worked for me is:
`auto esp32Led = JLed(jled::Esp32Hal(2, 7)).Blink(1000, 1000).Forever();`

jandelgado, 2020-11-18:
Yes, it's namespaced. I'll update the docs - thanks for the hint.

jandelgado, 2020-11-20:
README.md updated on b0373f90474701d95b667ed6a645efd7765e0e85 (master)

closed 2020-12-27 by jandelgado