jandelgado/jled #64: How to switch LED's current effect inside loop()

StarterCraft, 2020-12-25:
If I try writing code without dynamic LED effect switching like it's written below, it works normally:
```c++
#include "jled.h"
// do the parentheses actually break the library import?

#define myLED 13
JLed led(myLED);
//JLed led = JLed(myLed).On() - this does work too, but I don't see the difference

void setup() {
    led.On();
}

void loop() {
    led.Update();
}
```
But if I want to switch the LED's effect at the runtime according to a condition, nothing happens, the LED just ignores the instructions:
```C++
#include "jled.h"

JLed led = JLed(13).On().DelayAfter(2000);
bool _st;

void setup() {}

void loop() {
  if (someCondition) _st = true; //for example, switching the effect using a button
  if (_st) led.Blink(360, 360);
  led.Update();
}
```
How do I get rid of this problem?

jandelgado, 2020-12-25:
You need to reset the `_st` flag in you example, otherwise `led.Blink(360,360)` will be called over and over again, resetting the effect again and again. In your code e.g. you could do:

```c++
void loop() {
  if (someCondition) _st = true; //for example, switching the effect using a button
  if (_st) { 
      led.Blink(360, 360).Forever();
      _st = false;
  }
  led.Update();
}
```

jandelgado, 2020-12-25:
You need make sure only to call `led.Blink(360,360)`once. In your example, it will be called over and over again, resetting the effect again and again. In your code e.g. you could do:

```c++
bool _st = false;

void loop() {
  if (someCondition && !_st) { 
      _st = true; 
      led.Blink(360, 360).Forever();
  }
  led.Update();
}
```

StarterCraft, 2020-12-26:
Thank you! It works! :>

closed 2020-12-26 by jandelgado