jandelgado/jled #65: Reset fuction not restarting code

sjpiper145, 2021-01-04:
Hi @jandelgado ,

I love what you've built here and how easy it is compared to other methods.  I'm making a project with a picture of a heart and breathing LEDs to show the electrical pulses in an anatomical heart.  

The pulses need to have precise timing which I was able to set up well.  When the code runs, however, using the Reset() function doesn't seem to let the code start again, it just stops playing after the first one.  See attached images and code I'm using.

[Here's a link to a video of the project. ](https://www.dropbox.com/s/xelkhq3joidu7nj/2021-01-04%2013.45.46.mp4?dl=0)

Any ideas appreciated, big thanks again for your hard work on this.

Code:
[JLED Blinking Heart.zip](https://github.com/jandelgado/jled/files/5762920/JLED.Blinking.Heart.zip)

Cheers,
Steph

jandelgado, 2021-01-10:
Hi, the heart looks great, nice piece of work!

`Reset()` resets the JLed object immeadiately. If you want to run the the sequence over and over again, replace the `Reset()` by `Forever()` or `Repeat(N)` where `N` is the number of desired repetitions. By default, an effect is only played once.

jandelgado, 2021-01-18:
@sjpiper145 could you solve the problem?

sjpiper145, 2021-01-19:
Hi @jandelgado, I think I didn't not describe my problem very well.  I need the repetitions to start from the start, including both the DelayBefore() and DelayAfter() timing to make sure the pattern plays in the same sequence each time.  Using either the Forever () or Repeat () functions only play the sequence again excluding the timing for DelayAfter(). The sequence plays correctly for the first time, then all lights turn on together, then the sequence plays in reverse.  Thanks for following up with me on this, any ideas appreciated.

jandelgado, 2021-01-19:
Ok, basically you need to restart the sequence over, after the last LED-effect finished? I've already a ticket in the backlog, since this feature was already requested (see https://github.com/jandelgado/jled/projects/1#card-29307388). I'll implement it in the coming days.

jandelgado, 2021-01-24:
Checkout out the latest JLed Release 4.6.0: I've adde `Repeat(n)` and `Forever()` methods to the `JLedSequence` class. Check  README.md and the `multiled` example for details.

sjpiper145, 2021-01-25:
Wow awesome!  Will let you know how it goes.  Is there any way I can shout you a coffee? Big thanks.

sjpiper145, 2021-02-08:
This works great, happy for you to close this issue :)

closed 2021-02-08 by jandelgado