jandelgado/jled #68: Forever() not working for sequences

diematte, 2021-02-02:
Hei @jandelgado,
thank you for this great library. After adapting the HAL-layer I am able to use it on an STM32F3 with CubeMX.
I was planning to use sequences for outputting status-blinkcodes. However, when I do this:
```
JLedSequence sequence(JLedSequence::eMode::SEQUENCE, leds);
sequence.Forever();
```
the sequence stops after the first iteration.

I think the issue might be the following line in `Update()`:
```
is_running_ = ++iteration_ < num_repetitions_ &&
                      num_repetitions_ != kRepeatForever;
```
Because  `num_repetitions_` is set to `kRepeatForever` by `Forever()`, the second half of the above statement and thereby `is_running` will always be `false`.

I modified the line as follows and this seems to be working.
```
is_running_ = ++iteration_ < num_repetitions_ ||
                      num_repetitions_ == kRepeatForever;
```
Or am I getting something wrong?

Kind regards
Christoph

jandelgado, 2021-02-02:
Thanks for the report, this is indeed a bug... The only case not covered by a unit test - damn it. I'll fix it asap

br
jan

jandelgado, 2021-02-02:
fixed with 4.6.1 - could you please test it, @diematte ?

diematte, 2021-02-03:
It works! :-)
I also tried to run a very short sequence for an extended amount of time, to see if there is any weird effect when `iteration_` wraps around. This works as well.

There is just one edge-case I can think of:
In the unlikely case that a user wants to run **exactly** 65535 repetitions, the sequence will instead run forever (this is the case for single LEDs as well, right?). I don't think it is really worth addressing in the code. But maybe it could be noted in the description of Repeat().

PR #69 (2021-02-02): forever for sequence stopping after one iteration
fixes #68
closed 2021-02-05 by jandelgado