jandelgado/jled #70: Arduino Mega Pin issue

sjpiper145, 2021-02-08:
Hi Jandelgado,

I seem to be having a different issue with using pins on the Arduino Mega.  The project works great on any PWM pins, but I can't seem to get anything running on the extra pins in the A1-8 and 22-53 pin ranges.  I had a good read of the similar issue posted about the ESP, and the alternative GPIO(?) numbers don't seem to work either.  I can get these pins to work by using the standard digitalWrite blink function, however.  Once again, thanks for all your hard work on this project. Any ideas on this are appreciated! 

Here is the code I'm using: 

#include <jled.h>

JLed leds[] = {

// Blue LEDs (Works great)

    JLed(12).Breathe(1000).DelayAfter(2900),
    JLed(11).Breathe(1000). DelayBefore(100). DelayAfter(2800). MaxBrightness(100),
    JLed(13).Breathe(1000). DelayBefore(200). DelayAfter(2700). MaxBrightness(100),
    JLed(10).Breathe(1000). DelayBefore(300). DelayAfter(2600). MaxBrightness(100),
    JLed(9).Breathe(1000). DelayBefore(500). DelayAfter(2400).  MaxBrightness(100), 
    JLed(8).Breathe(1000). DelayBefore(700). DelayAfter(2200).  MaxBrightness(100), 
    JLed(7).Breathe(1000). DelayBefore(900). DelayAfter(2000).  MaxBrightness(100), 
    JLed(6).Breathe(1000). DelayBefore(1100). DelayAfter(1800). MaxBrightness(100),
    JLed(5).Breathe(1000). DelayBefore(1300). DelayAfter(1600). MaxBrightness(100),
    JLed(4).Breathe(1000). DelayBefore(1500). DelayAfter(1400). MaxBrightness(100),
    JLed(3).Breathe(1000). DelayBefore(1700). DelayAfter(1200). MaxBrightness(100),
    JLed(2).Breathe(1000). DelayBefore(1900). DelayAfter(1000). MaxBrightness(100),
    
// White LEDs (doesn't work)

    JLed(A8).Blink (250, 250). MaxBrightness(100),
    JLed(14).Blink(250, 250). MaxBrightness(100),
    JLed(29).Blink(250, 250). MaxBrightness(100). Repeat(5),
    JLed(25).Blink(250, 250). MaxBrightness(100). Repeat(5),
    JLed(49).Blink(250, 250). MaxBrightness(100). Repeat(5),
    JLed(48).Blink(250, 250). MaxBrightness(100). Repeat(10),
    
};
auto sequence = JLedSequence(JLedSequence::eMode::PARALLEL, leds).Repeat(1000);

void setup() { }

void loop() {
   
    sequence.Update();

    
    delay(1);

}

Cheers,
Steph

sjpiper145, 2021-02-15:
It seems as if I've been able to 'force' some LEDs to turn on by calling 'On' before blink using the code below: 

    JLed(24).On().DelayBefore(1000),
    JLed(24).Blink(250, 250). MaxBrightness(100),

Strange, I'll keep testing to see if I can get to the bottom of it.

jandelgado, 2021-02-18:
Hi, JLed uses PWM to set set the brightness of a LED. The Mega seems to only support PWM on 15 of it's pins. Refer to the spec for more details on which pins are PWM enabled.

There is also a discussion on https://forum.arduino.cc/index.php?topic=28493.0

closed 2021-05-25 by github-actions[bot]