jandelgado/jled #72: adding MinBrightness function

MaxDiDo-rgb, 2021-02-17:
Hi, and thank you for your awesome library!
I'm an Arduino beginner, so I hope you'll help me in realizing my request;

my question: is it possible to add function "MinBrightness" in order to simulate a "more real" breathing effect (i.e without completely turning off a led)?
I tried to use UserFunc to set a "gaussian" function (as explained in this link:  https://makersportal.com/blog/2020/3/27/simple-breathing-led-in-arduino ) but without success...

jandelgado, 2021-02-18:
Yes, as you already figured out, you can implement a user defined brightness evaluator class. A simple example (tested on an ESP32) that modifies the original JLed breathe function, so the minumum brightness will be 7 instead of 0, would be:

```c++
#include <jled.h>

class UserEffect : public jled::BreatheBrightnessEvaluator {
    // inherit constructor
    using jled::BreatheBrightnessEvaluator::BreatheBrightnessEvaluator;

    uint8_t Eval(uint32_t t) const override {
        const auto val = BreatheBrightnessEvaluator::Eval(t);
        return (val | 7);
    }
};

UserEffect userEffect(2500);
auto led = JLed(2).UserFunc(&userEffect).Forever();

void setup() {}

void loop() {
    led.Update();
}
```

If you need a more sophisticated function, just place your calculation inside the `Eval` method. The function must calculate `brightness(t,period) -> [0,255]`.

MaxDiDo-rgb, 2021-02-18:
thank you very much, I will try as soon as possible

closed 2021-05-25 by github-actions[bot]