jandelgado/jled #75: DelayAfter value not cleared by Stop?

voidcreatrix, 2021-04-29:
Hi there, I was struggling to transition between effects other effects and Blink (On -> Blink, Off-> Blink, FadeOn -> Blink). The Blink on-time seemed correct, but the off-time was way too long. I created this small program to test a theory and thought I would share in case it's useful. 

The program should change effects every 10 seconds: 
FadeOn (1s ramp, 1s delay: 5 fades)
Blink(500,500) (10 blinks)
FadeOn (same as before)
Blink(500,500) (same as before)

When I run it, the fade works correctly, but the blink only blinks 5 times -- as if it was Blink(500,1500)

But the second blink is correct. The only difference is that, in the second call to Blink, I added DelayAfter(0).

Is it possible that the DelayAfter value, used in the Fade effect, is being carried over to the Blink effect?

For reference, I am using ESP32 DevKitC.

===============================

`#include <jled.h>
#define LED_PIN 15

auto led = JLed(LED_PIN).FadeOn(1000).DelayAfter(1000).Forever();

uint8_t state = 0;

void setup() {
  Serial.begin(115200);
}

void loop() 
{
    
  uint32_t now = millis();

  if( (now > 10000) && (state == 0) )
  {
    led.Stop();
    led.Blink(500,500).Forever();
    state++;
  }
  else if( (now > 20000) && (state == 1) )
  {
    led.Stop();
    led.FadeOn(1000).DelayAfter(1000).Forever();
    state++;
  }
  else if( (now > 30000) && (state == 2) )
  {
    led.Stop();
    led.Blink(500,500).DelayAfter(0).Forever();
    state++;
  }
  
  led.Update();

}`

jandelgado, 2021-04-29:
Yes, that's the expected behaviour:
- `Stop()` just stops the current effect and turns the LED off
- `Reset()` resets the state of the current effects so it will start over. The effect _configuration_ (effect type, delay before/after) are not affected

In your case you could instead of of reconfiguring the existing effect, which will not change e.g. the delay values, you could also do a re-assignment like `led = JLed(LED_PIN).Blink(500,500).Forever();`

voidcreatrix, 2021-04-29:
Ah. My question is not so much about the Stop method, as about how the DelayAfter property from the FadeOn effect persisted into the Blink effect, unless I explicitly set the Blink to have a DelayAfter of 0.

 So the DelayAfter setting is a property of the LED object, not a property of the FadeOn effect?

I was thinking of FadeOn as one "effect", and Blink as a different "effect", so it didn't make sense that the DelayAfter from one of them persisted to the other. But it sounds like you're referring to an "effect" as the collection of all properties of the led object. Is that right?

That makes it seem like transitioning between effects is best done with a reassignment, as you say -- otherwise, every property has to be explicitly reassigned.

Thanks for your help, and for this very helpful library!

jandelgado, 2021-05-01:
Yes, `DelayAfter`, `DelayBefore`, `LowActive`, `MaxBrightness` and `Repeat` are properties of the LED object. The "effect" with (`Blink`, `Breathe`, `FadeOn`, `FadeOff`, `Candle`) is also a property. These properties can be set individually. When an effect is set, it will restart at the beginning, but delays etc. are not affected.

voidcreatrix, 2021-05-01:
Thanks, got it. It seems like a lot of the questions arising here are about
making changes to the properties, conditional transitions, etc. Would you
consider adding a code example (you're welcome to modify mine above if
helpful) showing the difference between changing one property, and
reassigning the object? It might cut down on the repetitive questions.
Adding a section to the instructions explaining this distinction would be a
help to other new users as well, I think.

Best,
Mylène

voidcreatrix, 2021-05-04:
As suggested, I used re-assignmnents to transition between effects.

I was able to compile

`led = JLed(RedPin).Blink(500,500).Forever();`

But when I tried 
'led = JLed(RedPin).Blink(500,500).Forever().Update()', I got the following compile error:

> no match for 'operator=' (operand types are 'jled::JLed' and 'bool')

Are there any other restrictions on reassignments?

voidcreatrix, 2021-05-04:
One more observation that might be of use. I'm using a common-anode tri-color LED, so needed the LowActive feature (thanks for including that!).  Every time I use the reassignment as noted above, if the LED was previously off, I see a barely-visible flash of light. Is it possible that the constructor is causing the pin to default to low briefly (thus turning on the LED), before the "Low Active" setting is processed?

When I modified my code to change the properties rather than reassigning, the brief flash went away.

Thanks again!

jandelgado, 2021-05-04:
> As suggested, I used re-assignmnents to transition between effects.
>
> I was able to compile
>
> `led = JLed(RedPin).Blink(500,500).Forever();`
>
> But when I tried
> `led = [...]

Yes, that because `Forever()` returns a JLed Object whereas `Update()` returns a boolean. Since C++ is a typed language, you get a compile time error (note that this is a different thing then a "exception"). 

Have a look at the documentation for [Update](https://github.com/jandelgado/jled#update):

> Update
>
>Call Update() periodically to update the state of the LED. Update returns true if the effect is active, and false when it finished.

closed 2021-08-09 by github-actions[bot]