jandelgado/jled #82: Update MaxBrightness Dynamically?

RossAWaddell, 2021-12-08:
Apart from the incremental functions, is there a way to update the MaxBrightness value dynamically?

`
auto ledStrbLts = JLed(9).Breathe(110, 113, 110)
                    .MaxBrightness(200).DelayAfter(500).Forever();

void loop()  
{

// int newBrightness = random(20, 255);
//ledStrbLits.MaxBrightness(newBrightness); ??????
  ledStrbLts.Update();

}
`

jandelgado, 2021-12-09:
Did you try your code? I think 

```c++
#include <jled.h>

auto ledStrbLts = JLed(9).Breathe(110, 113, 110)
                    .MaxBrightness(200).DelayAfter(500).Forever();

void setup() {}

void loop()  {
  ledStrbLts.MaxBrightness(random(20, 255));
  ledStrbLts.Update();
}
```
should work.

RossAWaddell, 2021-12-09:
Genius! Thanks again!

RossAWaddell, 2021-12-09:
Is the Breathe method supposed to flicker like this? I wanted to test to see if I could view the brightness level changes with the above code on my Uno so I changed the code to:

```
#include <jled.h>

auto ledStrbLts = JLed(10).Breathe(500, 1000, 500)
                    .MaxBrightness(200).DelayAfter(500).Forever();

void setup() {}

void loop()  {
  ledStrbLts.MaxBrightness(random(7, 255));
  ledStrbLts.Update();
}
```

With this, there is a definite flicker.

https://user-images.githubusercontent.com/29708900/145437869-2a01f7cd-93a3-4e9e-8bc1-a3fd7ffaa81f.mp4

jandelgado, 2021-12-10:
The `loop()` function is executed over and over again with a very high frequency. There we call the `JLed::Update` method, which iteratively updates the LED's output according to the chosen effect. Because you are changing the `MaxBrightness` also over and over again, not synchronized with the `JLed` effect running, it is flickering.

You need something like this (`Serial` stuff just added for debugging):

```c++
auto ledStrbLts = JLed(10).Breathe(500, 1000, 500)
                    .MaxBrightness(200).DelayAfter(500);  // note: .Forever() removed

void setup() {
    Serial.begin(9600);
}

void loop()  {
  // Update() returns false when the effect is done 'playing'
  if (!ledStrbLts.Update()) {
    const auto brightness= random(20,255);
    Serial.print("New brightness: ");
    Serial.println(brightness);
    ledStrbLts.MaxBrightness(brightness);
    // start over
    ledStrbLts.Reset();
  }
}
```

RossAWaddell, 2021-12-10:
That did it - thanks heaps!

closed 2021-12-10 by RossAWaddell