jandelgado/jled #88: FadeOn and Stay Forever

oscgonfer, 2022-02-22:
Hi,

Thank you for this fantastic library. I am working with it on a SAMD21 M0 (Feather RFM69) with the following code:

```
#include <jled.h>

// JLED
#define LASER_PIN       10
#define FRONTLIGHT_PIN  11

JLed pwm_pins[] = {JLed(LASER_PIN).Breathe(255).Forever(), JLed(FRONTLIGHT_PIN).Off()};
JLedSequence sequence(JLedSequence::eMode::PARALLEL, pwm_pins);
```

One of the packets received on the radio, triggers the following:

```
void laser_fade_in_slow(int brightness) {
    if (brightness){
        pwm_pins[0].FadeOn(FADESLOW).MaxBrightness(brightness);
    } else {
        pwm_pins[0].FadeOff(FADESLOW);
    }
}
```

I am sure I am missing something, but I am unable to figure out how to make it stay on at the desired `brightness` level forever, only getting a loop with the same FadeOn. 

Could you give some guidance?

Thanks in advance

jandelgado, 2022-02-23:
How often is the `laser_fade_in_slow` function called?

oscgonfer, 2022-02-23:
On demand, i.e. triggered remotely, but never anywhere close to intervals shorter than `FADESLOW`, which is 3000ms. Basically the idea is to turn the "laser" on with a fade in but leave it on after that until another function makes it turn off.

jandelgado, 2022-02-24:
I think you need to change the code to:

```c++
void laser_fade_in_slow(int brightness) {
    if (brightness){
        pwm_pins[0].FadeOn(FADESLOW).MaxBrightness(brightness).Repeat(1);
    } else {
        pwm_pins[0].FadeOff(FADESLOW).Repeat(1);
    }
}

```

Otherwise, the effect for pwn_pins[0] will run  `Forever`, because it was initially set.

oscgonfer, 2022-02-26:
Okay... so say I do this for both pins:

```
#include <jled.h>

#define LASER_PIN       21
#define FRONTLIGHT_PIN  22
#define FADESLOW 3000

JLed pwm_pins[] = {JLed(LASER_PIN).Breathe(255).Forever(), JLed(FRONTLIGHT_PIN).Off()};

void laser_fade_in_slow(int brightness, int what) {
    if (brightness){
        Serial.print("Set to FadeOn with brightness=");
        Serial.println(brightness);
        pwm_pins[what].FadeOn(FADESLOW).MaxBrightness(brightness).Repeat(1).Reset();
    } else {
        Serial.println("Set to FadeOff");
        pwm_pins[what].FadeOff(FADESLOW).Repeat(1).Reset();
    }
}
```

Note that `pwm_pins[0]` starts with a `Breathe(255).Forever()`, while the other one doesn't. 

Then, when I call `laser_fade_in_slow(255, 1)`, it works perfectly (I can later on update with no issue), but with ﻿`laser_fade_in_slow(255, 0)` it makes the FadeOn on `LASER_PIN`, but afterwards I can't update it.

jandelgado, 2022-02-27:
I modified the example to call your `laser_fade_in_slow` function every 5s with random values. Works as expected, LEDs changing state continously (I am using an ESP32 for the  test). What exactly you mean by "I can't update it"? Could you pleas try the example?

```c++
#include <jled.h>

#define LASER_PIN 21
#define FRONTLIGHT_PIN 22
#define FADESLOW 3000

#define LASER_PIN 21
#define FRONTLIGHT_PIN 22
#define FADESLOW 3000

JLed pwm_pins[] = {JLed(LASER_PIN).Breathe(255).Forever(),
                   JLed(FRONTLIGHT_PIN).Off()};

void laser_fade_in_slow(int brightness, int what) {
    Serial.print("For LED=");
    Serial.print(what);
    if (brightness) {
        Serial.print(", set to FadeOn with brightness=");
        Serial.println(brightness);
        pwm_pins[what]
            .FadeOn(FADESLOW)
            .MaxBrightness(brightness)
            .Repeat(1)
            .Reset();
    } else {
        Serial.println(", set to FadeOff");
        pwm_pins[what].FadeOff(FADESLOW).Repeat(1).Reset();
    }
}

void setup() {
    Serial.begin(9600);
}

void loop() {
    static unsigned long last_change_time_ = 0;

    pwm_pins[0].Update();
    pwm_pins[1].Update();

    delay(1);

    // every 5s call laser_fade_in_slow with a new random brightness
    if (millis() - last_change_time_ > 5000) {
        last_change_time_ = millis();

        const auto brightness = random(0, 255);
        const auto what = random(0, 2);
        if (brightness < 100) {
            laser_fade_in_slow(0, what);
        } else {
            laser_fade_in_slow(brightness, what);
        }
    }
}
```

oscgonfer, 2022-03-16:
Hi!

Sorry for not answering for a while. I didn't have the time to test this part until now.

**TL;DR**
- your test code works perfectly, no issue
- the framework I am using is Arduino, board is a RF69HCW featherm0 board. I am also using `RadioHead` and [neopixelbus](https://github.com/Makuna/NeoPixelBus) for managing radio communications and controlling RGB ledstrips respectively.  The code for the project I am making is [here](https://github.com/oscgonfer/iron-skulls-mask/tree/master/radio/server)
- Issue is now solved, process for it is below

**Step 1.** Adding `Repeat(1)` to `pwm_pins[X].FadeOn(FADESLOW).MaxBrightness(brightness);` makes the animation only be triggered once, which accomplishes the "stay on forever" part. 

**Step 2.** Adding `Reset()` to `pwm_pins[X].FadeOn(FADESLOW).MaxBrightness(brightness).Repeat(1);` doesn't seem to solve the "unable to update sequence" afterwards (see below)

**Step 3.** I saw in your code that you perform `pwm_pins[0].Update();` and `pwm_pins[1].Update();` in the loop, instead of a `sequence.Update();` of a `JLedSequence sequence(JLedSequence::eMode::PARALLEL, pwm_pins);` that I had declared after the pwm pins (see [first comment](https://github.com/jandelgado/jled/issues/88#issue-1147267855)). This seems to solve the issue, and actually, seems like the `JLedSequence` is no longer necessary and I don't fully understand why. Eitherway, issue seems to be solved and code is updated... so many thanks for your support, your code, and your library ❤️

closed 2022-03-16 by oscgonfer