jandelgado/jled #90: Change effect works in setup() only

Mark-81, 2022-03-10:
I'm testing this code on an Arduino Leonardo:
```

#include <jled.h>

JLed leds[] = 
{
  JLed(9).Off(),
  JLed(10).Off()
};
JLedSequence sequence(JLedSequence::eMode::PARALLEL, leds);

void ProcessLine(char *line)
{
  leds[1].Blink(100, 10).MaxBrightness(255).Forever();
  Serial.print("done");
}

void setup()
{
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  digitalWrite(9, LOW);
  digitalWrite(10, LOW);

  Serial.begin(115200);
  while (!Serial);
  Serial.setTimeout(10);

  //leds[1].Blink(100, 10).MaxBrightness(255).Forever(); // this works
}

void loop()
{
  static char inBuf[64];

  if (Serial.available())
  {
    size_t size = Serial.readBytes(inBuf, 64);
    inBuf[size] = '\0';
    ProcessLine(inBuf);
  }
 
  sequence.Update();
  delay(1);
}
```

I'm expecting it begins blinking whenever it receives something on the serial line. The "done" message is printed out. But the LED remains off.
Instead, if I uncomment the line in `setup()` it begins blinking after the serial port is open.

Where is my mistake here?

Mark-81, 2022-03-10:
Even if the second issue I linked above is very similar to my question, apparently I solved removing the sequence.

jandelgado, 2022-03-11:
The `JLedSequence` `sequence` object has a state and in your setup, the sequence is in state "not running", because both `JLed` objects are already/initially `Off`. So further calls to `sequence.Update()` will have no effect. Try to call `sequence.Reset()` in your `ProcessLine` function.

Also the `pinMode` and `digitalWrite` calls are not needed, since already handled by `JLed`

Mark-81, 2022-03-11:
Great thanks. Is there any advantage to use a parallel `sequence` instead of handling both `JLed` objects separately (i.e. calling `.Update()` for each one) ?

jandelgado, 2022-03-11:
No, the `JLedSequence` is just for convenience if you have e.g. 10 LEDs to update

closed 2022-03-13 by Mark-81