jandelgado/jled #92: Candle to turn off after effect.

Tezzatron81, 2022-03-22:
Is there a way or an example of turning off the candle effect after its done the sequence. I have 5 leds and i can get the candle effect working but i noticed they stay on after the sequence is done.

jandelgado, 2022-03-22:
Could you please post a complete example sketch?

Tezzatron81, 2022-03-22:
include <jled.h>
#include <Noiasca_button.h>            // download library from http://werner.rothschopf.net/microcontroller/202202_tools_led_en.htm

Button buttonA {2, HIGH};                  // active high: button connects pin with GND
unsigned long count = 0;
const int led1 = 11;
const int led2 = 10;
const int led3 = 9;
const int led4 = 6;
const int led5 = 5;

JLed candle1 = JLed(led1);
JLed candle2 = JLed(led2);
JLed candle3 = JLed(led3);
JLed candle4 = JLed(led4);
JLed candle5 = JLed(led5);

//JLed leds[] = {
   //           candle1.Candle(8,120),
  //            candle1.Off()
//};

//JLedSequence candlestop(JLedSequence::eMode::SEQUENCE, leds);

void setup() {
  Serial.begin(9600);
  pinMode(led1, OUTPUT); // Make Digital Pin 9 an OUTPUT
  pinMode(led2, OUTPUT); // Make Digital Pin 9 an OUTPUT
  pinMode(led3, OUTPUT); // Make Digital Pin 9 an OUTPUT
  pinMode(led4, OUTPUT); // Make Digital Pin 9 an OUTPUT
  pinMode(led5, OUTPUT); // Make Digital Pin 9 an OUTPUT
  buttonA.begin();                     // you have to call the .begin() method
  
}

void loop() {
  
  if (buttonA.wasPressed())  // fire if button was pressed

  { Serial.println(count);
    (count++);
    if (count >= 10) {
      count = 0;
    }
    if (count == 0) {  digitalWrite(led1, LOW);
        digitalWrite(led2, LOW);
        digitalWrite(led3, LOW);
        digitalWrite(led4, LOW);
        digitalWrite(led5, LOW);
    }
    if (count == 1) { digitalWrite(led1, HIGH);
    }
    if (count == 2) {digitalWrite(led2, HIGH);
    }
    if (count == 3) {digitalWrite(led3, HIGH);
    }
    if (count == 4) {digitalWrite(led4, HIGH);
    }
    if (count == 5) {digitalWrite(led5, HIGH);
    }
    if (count == 6) {effect1();
    }
    if (count == 7) {candle();
    }
    if (count == 8) {
  }
    if (count == 9 ) {
    
    
    }
    if (count == 10) {
    }

  }
   candle1.Update();
   candle2.Update();
   candle3.Update();
   candle4.Update();
   candle5.Update();
   //candlestop.Update();
}

void effect1()
{
        digitalWrite(led1, LOW);
        digitalWrite(led2, HIGH);
        digitalWrite(led3, LOW);
        digitalWrite(led4, HIGH);
        digitalWrite(led5, HIGH);
}

void candle()
{             candle5 = JLed(led5).Candle(8, 120).Repeat(2);
              candle4 = JLed(led4).Candle(7, 120);  
              candle3 = JLed(led3).Candle(8, 120);  
              candle2 = JLed(led2).Candle(7, 120);  
              candle1 = JLed(led1).Candle(8,120); 
              
              
               
}

Tezzatron81, 2022-03-23:
Im not sure where to call "off"

Tezzatron81, 2022-03-23:
ive added a timer library to the sketch and a void effect to turn off all the leds but im not sure if this is the best option. I would still like to have them turn off by themselves

Tezzatron81, 2022-03-25:
im also having trouble figuring out how to call a sequence inside my sketch on say (count==8)

jandelgado, 2022-03-25:
I've assembled an example that shows how to switch through some effects by pressing a button (uses newest JLed Version 4.10.0).
After each effect finishes, the LED is turned off. Is this what you are looking for?

Please not that no calls to `pinOut` and `digitalWrite` are needed, since everything is handled by `JLed`.

```c++
#include <JC_Button.h>  // https://github.com/JChristensen/JC_Button
#include <jled.h>

// initial effect
auto led = JLed(2).FadeOff(1000).Repeat(1);

// the button is connected to pin 5 and GND (thus using
// pull-up and being inverted)
auto button = Button(5, 25, true /*pull-up*/, true /*inverted*/);

void setup()
{
    Serial.begin(9600);
    Serial.println("press button to cycle through effects")
    button.begin();
}

void loop()
{
    static auto count = 0;
    button.read();

    if (!led.Update()) {
        led.Off();
    }

    if (button.wasPressed()) {
        count = (count + 1) % 7;
        switch (count) {
        case 0:
            Serial.println("5s candle");
            led.Candle(8, 120, 5000);
            break;
        case 1:
            Serial.println("5x breathe");
            led.Breathe(1000).Repeat(5);
            break;
        case 2:
            Serial.println("5x fade on");
            led.FadeOn(1000).Repeat(5);
            break;
        case 3:
            Serial.println("5x fade off");
            led.FadeOff(1000).Repeat(5);
            break;
        case 4:
            Serial.println("5x blink");
            led.Blink(500, 250).Repeat(5);
            break;
        case 5:
            Serial.println("5s on");
            led.On(5000).Repeat(1);
            break;
        case 6:
            Serial.println("off");
            led.Off();
            break;
        }
    }
}
```

Tezzatron81, 2022-03-25:
Yes thats what i meant. Ill have a look shortly. Would playing a multi sequence be basically the same theory.

Tezzatron81, 2022-03-26:
thats fantastic thankyou so much, but is there a way to have the sequence with multiple effects run as a button count?

jandelgado, 2022-03-29:
Yes, with upcoming JLed version 4.11.0 you can do things like:

```c++
// JLed dynamic sequence demo 
// 
// Toggles between sequences using a push button.
// After a sequence is finished, leds are turned off by calling
// Stop() on the sequence object.
//
// setup:
//   LEDs are connected to GPIO 22 and 23.
//   Button is connected to GPIO 5 and GND.
// 
#include <JC_Button.h>  // https://github.com/JChristensen/JC_Button
#include <jled.h>

JLed leds1[] = {
	JLed(22).FadeOn(500).Repeat(10),
	JLed(23).Off(),
};

JLed leds2[] = {
	JLed(22).Off(1),
	JLed(23).Blink(500, 500).Repeat(10),
};

JLed leds3[] = {
	JLed(22).FadeOn(500).DelayAfter(500).Repeat(10),
	JLed(23).DelayBefore(500).FadeOn(500).DelayAfter(500).Repeat(10),
};

auto seq = JLedSequence(JLedSequence::eMode::PARALLEL, leds3);
auto button = Button(5, 25, true /*pull-up*/, true /*inverted*/);

void setup()
{
    Serial.begin(9600);
    Serial.println("press button to cycle through effects");
    button.begin();
}

void loop()
{
    static auto count = 0;
    button.read();

    if (!seq.Update()) {
        // turn leds off after sequence is done
        seq.Stop();
    }

    if (button.wasPressed()) {
        switch (count) {
        case 0:
            Serial.println("FX1");
            seq = JLedSequence(JLedSequence::eMode::PARALLEL, leds1);
            seq.Reset();
            break;
        case 1:
            Serial.println("FX2");
            seq = JLedSequence(JLedSequence::eMode::PARALLEL, leds2);
            seq.Reset();
            break;
        case 2:
            Serial.println("FX3");
            seq = JLedSequence(JLedSequence::eMode::PARALLEL, leds3);
            seq.Reset();
            break;
        }
        count = (count + 1) % 3;
    }
}

```

Tezzatron81, 2022-03-29:
Oh that is magnificent. Im sorry for all the questions, but i just started using libraries, and this one is by far one of the best. My black panther statue with lights and sounds will be amazi g because of this. Thankyou for all your help. Ill keep an eye out for the update.

closed 2022-10-02 by github-actions[bot]