jandelgado/jled #95: Multiple HALs

troky, 2022-05-11:
Is it possible to implement multiple HALs ont he same board/project?
For example, I have ESP32 board with PCA9635 I2C driver and I want to use them both using JLed library at the same time, e.g. some LEDs are connected to ESP32 pins, and some are connected to PCA9635.

I have created custom HAL class but I dont know how to instantiate both HALs at the same time.

jandelgado, 2022-05-15:
I'll provide an example showing how to drive built-in ESP32 PWM's and an PCA9685 driver in the next days.

troky, 2022-05-16:
Driving PCA9685 (creating new HAL) is not a problem. I want to have std::list<JLed> of different HAL types.

jandelgado, 2022-05-17:
Could you please provide an example what the exact problem is?

troky, 2022-05-17:
There is no (abstract) base class/interface that coul be used.
For example: I need std:: list<JLedBase*>() where list elemnts can be JLed<HalType1>, JLed<Haltype2>, etc... so I can cast JLed<Haltype1> to JLedBase and call JLedBase->On() function etc...

jandelgado, 2022-05-20:
There is intentionally no base class to keep the memory footprint low. But coincidentially I am working on a feature allowing to stuff different objects like `JLed` objects or `JLedSequence` objects recursively into a `JLedSequence`. I'll try to provide a branch that you can test in the next days.

troky, 2022-05-22:
Well, for test, I de-templated your TLed class and made it abstract adding two virtual functions: analogWrite() and millis() and now I can create derived class from public TLed and implement those two funcions just like you do it in separate hal class. Problem solved... now I can have std::list<TLed *>()

closed 2022-08-26 by github-actions[bot]