jandelgado/jled #96: Change PWM base frequency

Mark-81, 2022-05-19:
Is it possible to change the base frequency of the PWM? For a specific application I need to use a PWM with a base frequency of about 11800 Hz.

jandelgado, 2022-05-20:
For the Arduino platform it is not supported currently, because the underlying Arduino API used does not support it. The ESP32 platform allows to set a custom frequency.

But for a workaround, look for example [here](https://nerdytechy.com/how-to-change-the-pwm-frequency-of-arduino/) or [here](https://forum.arduino.cc/t/adjusting-pwm-frequency-on-leonardo/122733)  on how to change the PWM frequency when using an Arduino. This could be done in initially in the `setup` function. But be aware to NOT change the frequency of timers affecting the `millis` function, because that could probably have an effect on `JLed` and other libraries.

jandelgado, 2022-06-12:
@Mark-81, is the question answered?

Mark-81, 2022-06-12:
Yes, thanks!

closed 2022-09-16 by github-actions[bot]