jandelgado/jled #99: Minimum brightness

Welsyntoffie, 2022-08-27:
I am not finding any references to minimum brightness for the breathing effect.
Currently LED's turn off then fades on again. I would like to set a minimum brightness.
Is that possible?

jandelgado, 2022-08-28:
Not yet, will add the feature soon.

Welsyntoffie, 2022-08-29:
Thats great news, thank you.

sauresahne, 2022-11-11:
Just stumbled upon this through Hackaday,  Wonderful library, thanks. 
That enhancement sounds great. I am going to use it for a bike light.
 
To summarize the great-rear-light-blinking-debate:  
Opinion 1: A blinking rear light gives you more attention from other drivers. 
Opinion 2: A steady rear light helps other people in judging distance and your speed. Blinking interferes with that. If a blinking light is your only rear light, other drivers can see you, but not always locate you precisely.

So, yeah. Being able to set a constant led brightness with a blink or breath on top would be really great.

jandelgado, 2022-11-16:
Released with 4.12.0 https://github.com/jandelgado/jled/releases/tag/v4.12.0

Welsyntoffie, 2022-12-19:
Works great. Thank you.
I did notice that the stop command does not turn the led off but keeps it at the lowest set brightness. In my project I had to analogwrite=0 to completely turn off the led. But the MinBrightness does work great. Closing this issue

jandelgado, 2022-12-20:
Good Point. Will think about it over the holidays. I think it would still be good to be able to turn a LED fully off with Stop, perhaps by adding an optional parameter.

jandelgado, 2023-08-21:
fyi @Welsyntoffie: with JLed Release 4.13.0 (see #121 ) it is now possible to fully turn of the LED or keep the last value with `Stop()`.

Welsyntoffie, 2023-08-22:
Great. Thanks for info. Still going strong with my CSGO stage light strip.