jandelgado/jled #110: Make fade-resolution choosable

hansimglueck, 2023-01-07:
As a "follow-up" to my wish https://github.com/jandelgado/jled/issues/109, I also would like very much too see the option of 10bit, 12 bit or 16bit pwm-resolution in this beautiful Library.
In this case I am using ESP to drive a meter of LED-strip. The steps of the 8bit resolution are clearly visible in dark situations...

As I understand this might be tricky to implement in a platform-independent way, but there must be a way, isnt it?

Anyway thanks for the great work!

rainyotter, 2023-03-24:
Seconding this, 8-bit vs 10-bit is extremely noticeable when fading leds at dimmer brightness.

jandelgado, 2023-03-25:
@hansimglueck @rainyotter could you provide an example which I can use as the base for some research please?

Welsyntoffie, 2023-03-26:
I am fading 30meters of white LED rope lights using standard library and see no noticeable weird effects. Fading is smooth.
I fade from 1 to 255 using Arduino Mega.

rainyotter, 2023-03-26:
I use a LED bulb as a 'sunrise' light, where it slowly dims up over several minutes. At 8-bit resolution, the steps are extremely noticeable at the low end. Even dimming by hand (i.e. with a potentiometer) they're noticeable. I use 10-bit now (not using this library), but even that still has some visible steps during slow fades.