jandelgado/jled #117: Question/Feature: Get current brightness

arnolddej, 2023-07-19:
Currently building a cabinet light using this library. Absolutely fantastic. 
But I have one question. Is it possible to get the current brightness level.

It's not a first world problem, but whenever the doors of the cabinet opens/closes too fast it start at the max or min brightness level while its for example only half way. I'm using only FadeOn and FadeOff functions btw. When I could get the current brightness this can be fixed using the Fade function.

jandelgado, 2023-07-29:
The "current" brightness level is a function of time. JLed does not store the lastly written level, since this would require an additional byte in the internal state. I'll check if it's possible to return the current level with the `Update` method.

jandelgado, 2023-07-30:
@arnolddej, On the `return_current_level_from_update` branch  I added experimental support for the `Update()` method returning also the brightness level written, while keeping compatibility with old code. Have a look at the new `UpdateResult` class. Use it as follows:

```c++
auto result = led.Update();    // returns now an `UpdateResult` object that can be cast to a `bool`

// signals that a new value was written out to the LED
if (result.WasChanged()) {
    auto level = result.Value();
    // do something with the level ....
}
```

arnolddej, 2023-07-30:
Thanks! I will try it asap.

arnolddej, 2023-08-03:
@jandelgado I've tested it. 

It seems like the value not always correspond to the actual brightness whenever you interrupt the running Fade() command with another Fade() command. Also whenever you interrupt the running cmd it sometimes will start at the minimum/maximum brightness level  while its current value is only half for example.

Is it even possible to call Fade() with another Fade() on the same object while the first cmd isn't finished yet.
 
My current turn on/off code: 

```C++
void turn_on() {
if (ledstrip1.IsRunning()) {
   ledstrip1.Stop();
}
ledstrip1.Fade(current_value_ledstrip1, 255, DURATION);
}

void turn_off() {
if (ledstrip1.IsRunning()) {
   ledstrip1.Stop();
}
ledstrip1.Fade(current_value_ledstrip1, 0, DURATION);
}
```

aGGreSSiv, 2023-08-04:
Hello. I have the same problem. I follow your correspondence silently. :)

aGGreSSiv, 2023-08-04:
Since I am not a software developer, it is difficult for me to control 15 LED strips connected to different GPIOs of ESp32. For better or worse, I managed to do this, but for a programmer, the long code below could have been much shorter.

`
// https://github.com/jandelgado/jled
#include <WiFi.h>
#include <WiFiClient.h>
#include <BlynkSimpleEsp32.h>
#include <TimeLib.h>
#include <WidgetRTC.h>
#include <jled.h>

auto led1 = JLed(12);
auto led2 = JLed(13);
auto led3 = JLed(14);
auto led4 = JLed(25);
auto led5 = JLed(26);
auto led6 = JLed(27);
auto led7 = JLed(32);
auto led8 = JLed(33);
auto led9 = JLed(19);
auto led10 = JLed(23);
auto led11 = JLed(18);
auto led12 = JLed(5);
auto led13 = JLed(17);
auto led14 = JLed(16);
auto led15 = JLed(4);

bool flagState = false;
bool virtualState = false;

String newHostname = "Deneme";
const char* ssid = "barisea";
const char* pass = "66376637bd";
int ReCnctFlag;       // Reconnection Flag
int ReCnctCount = 0;  // Reconnection counter

bool isFirstConnect = true;
char auth[] = "_yy8K-NNbnGzH-LBf5aihJ28ZlqVV7Hl";
//char server[] = "brsglr.mooo.com";  // IP for Local Cloud Server
//int port = 80;
char server[] = "192.168.1.2";  // IP for Local Cloud Server
int port = 8080;

int sensorPin = 15;
int sensorNewState;
bool upTrigger = false;
bool downTrigger = false;

int maxB = 255;
int minB = 0;

BlynkTimer timer;
WidgetRTC rtc;

unsigned long previousMillis = 0;
unsigned long currentMillis = 0;
long interval = 9000;

BLYNK_WRITE(V27) {
  maxB = param.asInt();
}

BLYNK_WRITE(V28) {
  minB = param.asInt();
}

BLYNK_WRITE(V25) {

  virtualState = param.asInt();
  if (virtualState == 1 & flagState == 0) {  // Run only if Switch ON and Flag set to 0
    flagState = 1;                           // Set flag to prevent multiple occurrences.
    led1.Fade(0, maxB, 200).DelayBefore(100);
    led2.Fade(0, maxB, 200).DelayBefore(200);
    led3.Fade(0, maxB, 200).DelayBefore(300);
    led4.Fade(0, maxB, 200).DelayBefore(400);
    led5.Fade(0, maxB, 200).DelayBefore(500);
    led6.Fade(0, maxB, 200).DelayBefore(600);
    led7.Fade(0, maxB, 200).DelayBefore(700);
    led8.Fade(0, maxB, 200).DelayBefore(800);
    led9.Fade(0, maxB, 200).DelayBefore(900);
    led10.Fade(0, maxB, 200).DelayBefore(1000);
    led11.Fade(0, maxB, 200).DelayBefore(1100);
    led12.Fade(0, maxB, 200).DelayBefore(1200);
    led13.Fade(0, maxB, 200).DelayBefore(1300);
    led14.Fade(0, maxB, 200).DelayBefore(1400);
    led15.Fade(0, maxB, 200).DelayBefore(1500);
    flagState = 0;  // Reset flag
    downTrigger = true;
  } else {
    led1.Fade(200, minB, 200).DelayBefore(1500);  // en üst ve en alt basamaklarda ışığı azıcık açık bırak. yürüyen önünü görsün
    led2.Fade(200, minB, 200).DelayBefore(1400);  // en üst ve en alt basamaklarda ışığı azıcık açık bırak. yürüyen önünü görsün
    led3.Fade(200, minB, 200).DelayBefore(1300);
    led4.Fade(200, minB, 200).DelayBefore(1200);
    led5.Fade(200, minB, 200).DelayBefore(1100);
    led6.Fade(200, minB, 200).DelayBefore(1000);
    led7.Fade(200, minB, 200).DelayBefore(900);
    led8.Fade(200, minB, 200).DelayBefore(800);
    led9.Fade(200, minB, 200).DelayBefore(700);
    led10.Fade(200, minB, 200).DelayBefore(600);
    led11.Fade(200, minB, 200).DelayBefore(500);
    led12.Fade(200, minB, 200).DelayBefore(400);
    led13.Fade(200, minB, 200).DelayBefore(300);
    led14.Fade(200, minB, 200).DelayBefore(200);
    led15.Fade(200, minB, 200).DelayBefore(100);
    downTrigger = false;
  }
}

BLYNK_WRITE(V26) {
  if (param.asInt()) {
  }
}

BLYNK_CONNECTED() {  //Start Blynk connection
  rtc.begin();
  ReCnctCount = 0;
  if (isFirstConnect) {
    //rtc.begin();
    Blynk.syncAll();
    Serial.println("Blynk.synced");
    isFirstConnect = false;
  }
}

void setup() {
  Serial.begin(115200);
  WiFi.hostname(newHostname.c_str());
  WiFi.begin(ssid, pass);
  Blynk.config(auth, server, port);
  Blynk.connect();
  setSyncInterval(60);
  timer.setInterval(100L, checkLights);
  attachInterrupt(sensorPin, sensorButtonPressed, RISING);
}

void sensorButtonPressed() {
  Serial.println("sensorButtonPressed");
  int sensorNewState = digitalRead(sensorPin);
  if (sensorNewState == HIGH) {
    previousMillis = currentMillis;
    if (downTrigger == false) {
      upTrigger = true;
    }
  }
  return;
}

void loop() {
  timer.run();
  currentMillis = millis();
  //sensorNewState = digitalRead(sensorPin);

  led1.Update();
  led2.Update();
  led3.Update();
  led4.Update();
  led5.Update();
  led6.Update();
  led7.Update();
  led8.Update();
  led9.Update();
  led10.Update();
  led11.Update();
  led12.Update();
  led13.Update();
  led14.Update();
  led15.Update();

  if (sensorNewState == HIGH) {
  }
  
  if (upTrigger == true) {
    Blynk.virtualWrite(V25, 1);
    Blynk.syncVirtual(V25);
    upTrigger = false;
  }

  if ((currentMillis - previousMillis >= interval) && (downTrigger == true)) {
    Blynk.virtualWrite(V25, 0);
    Blynk.syncVirtual(V25);
  }

  if (Blynk.connected()) {  // If connected run as normal
    Blynk.run();
  } else if (ReCnctFlag == 0) {  // If NOT connected and not already trying to reconnect, set timer to try to reconnect in 30 seconds
    WiFi.disconnect();
    WiFi.hostname(newHostname.c_str());
    WiFi.begin(ssid, pass);
    ReCnctFlag = 1;  // Set reconnection Flag
    Serial.println("Starting reconnection timer in 30 seconds...");
    timer.setTimeout(5000L, []() {  // Lambda Reconnection Timer Function
      ReCnctFlag = 0;               // Reset reconnection Flag
      ReCnctCount++;                // Increment reconnection Counter
      Serial.print("Attempting reconnection #");
      Serial.println(ReCnctCount);
      Blynk.connect();  // Try to reconnect to the server
    });                 // END Timer Function
  }
}

void checkLights() {
}
`

jandelgado, 2023-08-05:
@arnolddej 
Could you please post a complete sketch. Especially the part where current_value_ledstrip1 is obtained is of interest here

jandelgado, 2023-08-06:
@aGGreSSiv please open another issue with your question and don't post code with secrets!

arnolddej, 2023-08-06:
> @arnolddej 
> Could you please post a complete sketch. Especially the part where current_value_ledstrip1 is obtained is of interest here
> 

I am currently on holiday for a week. Will share the code when I'm back. Basically, I assign a global ```uint8_t``` variable like your example. I then use this variable in the ```Fade()``` function.

jandelgado, 2023-08-07:
No problem - me too :-) Anyway, I think i found the problem in the meantime ...

jandelgado, 2023-08-18:
@arnolddej I did a few changes, please test again.

arnolddej, 2023-08-23:
@jandelgado Unfortunately I already had to build in the lighting. Will test the circuit on a breadboard soon.

rovale, 2024-01-28:
I am happy you started this feature! I want to use it to synchronize the state of an RGB led.

Currently I test it with a Blink Forever. Right now WasChanged always returns true.

SaKiEQ, 2024-11-07:
Could you elaborate a bit more. I do understand the concept outlined above, yet I do not see how you implemented the update() return. 
I am facing the same issues where I need the approximate current duty level. 
I need to update a web interface with the current levels of PWM channels via UDP, hence I can not just send levels from the program it self when a pwm.fadeOn() is called,  I need the current duty cycle that is being calculated. 

I am working on a ESP32 so an array of uint's for the duty cycles * 16 channel is not really a memory constraint. 

Any example possible ?

jandelgado, 2024-11-10:
Look at #131 where I (re-)implemented the feature. You can now pass an optional pointer to `Update`  which will return the last brightness value, if any.