jandelgado/jled #122: Repeat() and DelayAfter() together in a sequence

mattbk, 2023-09-09:
When I run code like this, I would expect the first effect to repeat three times with a delay of three seconds after each time. But the first effect runs once. Am I missing something?

Modified from https://github.com/jandelgado/jled/blob/master/examples/sequence/sequence.ino.
```
#include <jled.h>

constexpr auto LED_PIN = 3;

JLed leds[] = {
    JLed(LED_PIN).Breathe(2000).Repeat(3).DelayAfter(1000),
    JLed(LED_PIN).Blink(750, 250).Repeat(3),
    JLed(LED_PIN).FadeOff(1000).Repeat(3),
    JLed(LED_PIN).Blink(500, 500).Repeat(3),
    JLed(LED_PIN).FadeOn(1000).Repeat(3),
    JLed(LED_PIN).Off(5000)
};

JLedSequence sequence(JLedSequence::eMode::SEQUENCE, leds);

void setup() { }

void loop() {
    sequence.Update();
    delay(1);
}
```

Thanks a bunch for the Morse example, I'm using it for exactly that purpose (to key a radio).

mattbk, 2023-09-09:
I realized that for my use case, I can add an extra space (" ") at the end of the `MorseEffect` message, and the messages won't run together. I don't think this manual intervention is possible for regular LED sequences, though.

jandelgado, 2023-09-10:
Thanks for the report - I can confirm that there is a problem in the `JLedSequence` code. Will investigate further ...

jandelgado, 2023-09-10:
fixed with 4.13.1, please check if it's ok now

mattbk, 2023-09-12:
Works a treat, thank you!

closed 2023-09-12 by mattbk