jandelgado/jled #126: Update() not returning false for On()

danyhm, 2023-11-02:
Hello,

if the following code:
```
auto ledOn = JLed(LED).On().DelayAfter(500);

loop()
{
if(!ledOn.update())
{
// code here
}
}
```

Update() is not returning false. is this by design for solid states ?

jandelgado, 2023-11-02:
Yes, that's by design an also documented, see: https://github.com/jandelgado/jled#static-on-and-off

> Technically, Set, On and Off are effects with a default period of 1ms, that set the brightness to a constant value. Specifying a different period has an effect on when the Update() method will be done updating the effect and return false (like for any other effects). This is important when for example in a JLedSequence the LED should stay on for a given amount of time.

In your example `Update` will return `true` for (1+500)ms == 501ms

danyhm, 2023-11-02:
Well it's always returning true. So i take it this should not happen?

jandelgado, 2023-11-03:
Could you please provide a complete example so I can try to reproduce the problem? Which version of JLed are you using?

danyhm, 2023-11-03:
i'm sorry. you were right. it does return false. my bad

closed 2023-11-03 by danyhm