jandelgado/jled #127: Delay only working for low values

BjornB2, 2023-11-13:
Hey Jan,

I'm using the following function:

      led[i] = JLedPCA9685(i).Breathe(random(minFadeIn, maxFadeIn), random(minOn, maxOn), random(minFadeOut, maxFadeOut)).DelayBefore(random(minOff, maxOff));

It all works fine for lower values of minOff and maxOff, (10-30 sec), but if I enter 120000 and 180000 ms it goes much faster. Could it be that the max length of the value is limited or something? With those higher values the first leds come on right away, while a minimum of 120 sec is expected.

Thanks!

Björn

BjornB2, 2023-11-13:
I think I found it. It seems uint16_t is used to store the value for the delay, and if I am correct uint16_t has a max value of 65535.

jandelgado, 2023-11-19:
Yes, that is the reason. To keep the memory footprint small, time values are modelled as `uint16_t` currently. However I plan to make the type a c++ template parameter, so, if needed, the type can be changed to e.g. `uint32_t`

closed 2024-02-24 by github-actions[bot]