jandelgado/jled #128: LowActive() causes weird Fade behaviour

RossAWaddell, 2024-07-24:
I need to connect the -ve pin of LEDs to the Arduino as I'm using a ULN2803A transistor to power 10 5mm LEDs. As this is the opposite of the usual single LED connection I'm using the LowActive() method to invert the fade, but it does a weird _**blip**_ at the beginning which I don't see when connecting the +ve LED lead to the Arduino pin.

Any idea what's happening here? Ideally I'd like a smooth fade from off to the max brightness value.

**+ve LED Lead Connected to Arduino Pin**

`#include <jled.h>

#define AMBER_LED_PIN 3
#define AMBER_LED_FADE_UP_TIME 4000// Time to power up (power first turned on)
#define PULSE_TIME 1200            // Fade down/up pulse time
#define MAX_BRIGHT_AMBER_LEDS 220
#define MIN_BRIGHT_AMBER_LEDS 110

JLed leds[] = {
JLed(AMBER_LED_PIN).FadeOn(AMBER_LED_FADE_UP_TIME).MaxBrightness(MAX_BRIGHT_AMBER_LEDS).DelayAfter(250),
    JLed(AMBER_LED_PIN).Fade(MAX_BRIGHT_AMBER_LEDS, MIN_BRIGHT_AMBER_LEDS, PULSE_TIME/2),
    JLed(AMBER_LED_PIN).Breathe(PULSE_TIME).MaxBrightness(MAX_BRIGHT_AMBER_LEDS)
      .MinBrightness(MIN_BRIGHT_AMBER_LEDS).DelayAfter(250).Forever()
};

JLedSequence amberLEDseq(JLedSequence::eMode::SEQUENCE, leds);

void setup() { }

void loop() {
    amberLEDseq.Update();
}
`

[YouTube video](https://youtu.be/WjzOhVDtfPQ)

**-ve LED Lead Connected to Arduino Pin w/ LowActive()**

`#include <jled.h>

#define AMBER_LED_PIN 3
#define AMBER_LED_FADE_UP_TIME 4000// Time to power up (power first turned on)
#define PULSE_TIME 1200            // Fade down/up pulse time
#define MAX_BRIGHT_AMBER_LEDS 220
#define MIN_BRIGHT_AMBER_LEDS 110

JLed leds[] = {
JLed(AMBER_LED_PIN).LowActive().FadeOn(AMBER_LED_FADE_UP_TIME).MaxBrightness(MAX_BRIGHT_AMBER_LEDS).DelayAfter(250),
    JLed(AMBER_LED_PIN).LowActive().Fade(MAX_BRIGHT_AMBER_LEDS, MIN_BRIGHT_AMBER_LEDS, PULSE_TIME/2),
    JLed(AMBER_LED_PIN).LowActive().Breathe(PULSE_TIME).MaxBrightness(MAX_BRIGHT_AMBER_LEDS)
      .MinBrightness(MIN_BRIGHT_AMBER_LEDS).DelayAfter(250).Forever()
};

JLedSequence amberLEDseq(JLedSequence::eMode::SEQUENCE, leds);

void setup() { }

void loop() {
    amberLEDseq.Update();
}
`

[YouTube video](https://youtu.be/BRnIPiKpTV0)

jandelgado, 2024-08-20:
Hi @RossAWaddell, 
* which Arduino model are you using?
* Is there a difference if you are not using the transistor and just driving a single LED?
* could you please provide the circuit so I can try to re-build it?

RossAWaddell, 2024-08-20:
Hi @jandelgado ,

- Arduino Uno
- If I use a Darlington transistor (e.g. ULN2803A) then I don't need to use LowActive() and everything works perfectly. But if I use just a NPN transistor, then it behaves the same as when driving a single LED
- See Fritzing schematic below

[image]
[image]

jandelgado, 2024-08-31:
Hi @RossAWaddell , 

I did some testing, and it seems that it has to do with the GPIO port initialization (`pinMode()`) during which the Arduino SDK will set the level to 0, wich causes the visible `blip` when using a low-active connection. As a workaround, try the following:

```c++

void setup() { 
  // add this line
  ::digitalWrite(AMBER_LED_PIN, HIGH);
}
```

RossAWaddell, 2024-08-31:
Thanks! I'll give it a try.

jandelgado, 2024-09-05:
I did some more tests. What probably helps in your case is to add a `delay(25)` to the main `loop()`. In my tests that eliminated the blip on startup.

closed 2024-12-10 by github-actions[bot]